mani-skill/ManiSkill #514: How to handle unexpected motions?

HenryWJL, 2024-08-21:
https://github.com/user-attachments/assets/83caf8be-0482-4202-8588-52efd2881de1

Hi, I added a flying leaphand into the environment, and I used delta controllers to control hand root pose and finger joint positions. In the attached video, I only moved one joint and kept others still, but I found that when the joint position exceeded its limits, the whole body would also move accordingly. How can I fix this issue?

StoneT2000, 2024-08-21:
can you share your agent class that inherits base agent that defines this robot?

HenryWJL, 2024-08-21:
Here is the agent class:
```python
from copy import deepcopy
from typing import List

import numpy as np
import sapien
import torch

from mani_skill import PACKAGE_ASSET_DIR
from mani_skill.agents.base_agent import BaseAgent, Keyframe
from mani_skill.agents.controllers import *
from mani_skill.agents.registration import register_agent
from mani_skill.utils import sapien_utils
from mani_skill.utils.structs.pose import vectorize_pose

@register_agent()
class LeapHandRight(BaseAgent):
    uid = "leaphand_right"
    urdf_path = f"{PACKAGE_ASSET_DIR}/robots/leaphand/leap_hand_right_glb.urdf"
    urdf_config = dict(
        _materials=dict(
            tip=dict(static_friction=2.0, dynamic_friction=1.0, restitution=0.0)
        ),
        link={
            "fingertip": dict(
                material="tip", patch_radius=0.1, min_patch_radius=0.1
            ),
            "fingertip_2": dict(
                material="tip", patch_radius=0.1, min_patch_radius=0.1
            ),
            "fingertip_3": dict(
                material="tip", patch_radius=0.1, min_patch_radius=0.1
            ),
            "thumb_fingertip": dict(
                material="tip", patch_radius=0.1, min_patch_radius=0.1
            ),
        },
    )
    keyframes = dict(
        open_facing_up=Keyframe(
            qpos=[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], pose=sapien.Pose(p=[0, 0, 0.5])
        ),
        open_facing_side=Keyframe(
            qpos=[0, 0, 0, 0, np.pi / 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], pose=sapien.Pose(p=[0, 0, 0.5])
        ),
        open_facing_down=Keyframe(
            qpos=[0, 0, 0, 0, np.pi, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], pose=sapien.Pose(p=[0, 0, 0.5])
        ),
    )

    def __init__(self, *args, **kwargs):
        self.joint_stiffness = 400
        self.joint_damping = 10
        self.joint_force_limit = 50

        self.palm_joint_names = [
            "base_joint_x",
            "base_joint_y",
            "base_joint_z",
            "base_joint_roll",
            "base_joint_pitch",
            "base_joint_yaw",
        ]
        self.finger_joint_names = [
            "0",
            "1",
            "2",
            "3",
            "4",
[... 990 lines cut ...]
  <link name="middle_tip_head"/>
  <link name="ring_tip_head"/>
  <joint name="thumb_tip" type="fixed">
    <parent link="thumb_fingertip"/>
    <child link="thumb_tip_head"/>
    <origin rpy="0 0 0" xyz="0 -0.06 -0.015"/>
  </joint>
  <joint name="index_tip" type="fixed">
    <parent link="fingertip"/>
    <child link="index_tip_head"/>
    <origin rpy="0 0 0" xyz="0 -0.048 0.015"/>
  </joint>
  <joint name="middle_tip" type="fixed">
    <parent link="fingertip_2"/>
    <child link="middle_tip_head"/>
    <origin rpy="0 0 0" xyz="0 -0.048 0.015"/>
  </joint>
  <joint name="ring_tip" type="fixed">
    <parent link="fingertip_3"/>
    <child link="ring_tip_head"/>
    <origin rpy="0 0 0" xyz="0 -0.048 0.015"/>
  </joint>
</robot>

```

StoneT2000, 2024-08-22:
I see. It seems like a reaction force from the block move the hand. What actions did you send?

HenryWJL, 2024-08-23:
Here are the actions:
```python
actions = torch.zeros(num_envs, 22)
actions[:, 18] = 0.2
```
As I utilized delta control, there should not be any movement on the whole body except the thumb.

HeegerGao, 2024-08-23:
> I see. It seems like a reaction force from the block moves the hand. What actions did you send?

No. The finger never touches the block. The thumb reaches its limit, and then it drives the whole hand to rotate. You can see other finger joints also moved. It seems like the whole hand suddenly becomes loose after the thumb reaches the limit.

StoneT2000, 2024-08-23:
@HeegerGao oh this makes much more sense. That seems the more likely reason the hand rotation occurred.

A "fix" is to modify the hand controller such that you are using delta actions for the finger but absolute actions for the root xyz rpy controls. This should force the hand to remain in a specific pose while still letting you move the hand around if you want

If you really want to fix the hand in place, remove the root xyz rpy joints and links

HenryWJL, 2024-08-24:
Unfortunately, the pd_ee_pose controller does not support absolute control in gpu sim, and since we are using a flying hand, we do not want to "fix the hand in place". The problem seems to lie in physx, but we have not found a solution till now.

StoneT2000, 2024-08-24:
You can do absolute joint position control for the root xyz rpy, that's the same thing as pd ee pose actually in this particular case. Use the pd joint position controller configuration and set use_delta=False and normalize_actions=False. The 6 action dimensions corresponding to that controller can be used to directly set where the robot hand is and try to keep it there.

HenryWJL, 2024-08-24:
Hi, I modified my code like this:
```python
palm_pose = PDJointPosControllerConfig(
    self.palm_joint_names,
    lower=-1,
    upper=1,
    stiffness=self.joint_stiffness,
    damping=self.joint_damping,
    force_limit=self.joint_force_limit,
    use_delta=False,
    normalize_action=False,
)
```
and set the actions like this:
```python
actions = torch.zeros(num_envs, 22)
actions[:, 18] = 0.2
```
It seemed to work.

https://github.com/user-attachments/assets/3bc6bb65-a884-45b7-ad21-3eec0d6cf159

But when I tested other joints, the problem still existed. For instance, I set the actions like this:
```python
actions = torch.zeros(num_envs, 22)
actions[:, 10] = 0.2
```
and when the moving finger hit its nearby finger, all the fingers began to move strangely.

https://github.com/user-attachments/assets/a2b41911-9ba9-44b0-9dcf-4ce49f2cdf81

Finally, I also replaced the delta controller for finger joints with an absolute controller, and the problem seemed to be completely solved. Anyway, thanks for your guidance, and I hope the problem of delta controllers can be solved in the future.

HenryWJL, 2024-08-24:
By the way, there must be something wrong with the environment, not just the controllers. Even if I utilized absolute controllers, the same problem would arise if I passed (current_qpos + delta_qpos) to the controllers.

StoneT2000, 2024-08-28:
@HenryWJL it could be a joint limit issue? Are the joint limits too big and unrealistic?

The fact a finger can hit another finger in the video above seems like a poor robot definition. Either disable some self collisions or lower the joint limits

HenryWJL, 2024-08-30:
Hey, reducing the joint force limits seemed to work! Thanks a lot.

closed 2024-08-30 by StoneT2000