mani-skill/ManiSkill #520: [Question] `max_episode_steps` for `num_envs>1`

hesic73, 2024-08-22:
The source code for the `step` function of  `gymnasium.wrappers.time_limt.TimeLimit` is [as follows](https://github.com/openai/gym/blob/dcd185843a62953e27c2d54dc8c2d647d604b635/gym/wrappers/time_limit.py#L39):

```python
def step(self, action):
        """Steps through the environment and if the number of steps elapsed exceeds ``max_episode_steps`` then truncate.

        Args:
            action: The environment step action

        Returns:
            The environment step ``(observation, reward, terminated, truncated, info)`` with `truncated=True`
            if the number of steps elapsed >= max episode steps

        """
        observation, reward, terminated, truncated, info = self.env.step(action)
        self._elapsed_steps += 1

        if self._elapsed_steps >= self._max_episode_steps:
            truncated = True

        return observation, reward, terminated, truncated, info
```

This implementation seems to assume a single environment where truncated and terminated are booleans. However, in ManiSkill, when `num_envs > 1`, this assumption doesn't hold. I've observed that **the `TimeLimit` wrapper is enabled** by default if `max_episode_steps` is set during registration.

 I printed out `self._elapsed_steps ` and it seems to remain at one most of the time, and never exceeds `max_episode_steps`  (in my limited observation).

My questions are:

- How does `max_episode_steps` behave when `num_envs > 1`?
- If I want each single environment to have a time limit and ensure **they are not wrongly truncated**, how can I achieve this in ManiSkill? For example, I can track the episode steps and check it in `evaluate`:

```python
fail = torch.logical_or(
            palm_object_dist > 0.5,  self._episode_steps >= self.max_episode_steps).squeeze(-1) * (~success)  # (n,)
```

StoneT2000, 2024-08-22:
I think I fixed this issue from a while back. By default, the default timelimit wrapper (CPU 1 env version) is always replaced by our own batched version of it. You can see that the truncated values returned by env.step are always a batched torch tensor and are all True if max episode steps is reached. The behavior is the same with any value of `num_envs`. `truncated` is always a torch tensor of `num_envs` elements

hesic73, 2024-08-22:
See this example:

```python
import torch

import gymnasium as gym

from mani_skill.envs import BaseEnv

from mani_skill.utils.registration import register_env
import time

_MAX_EPISODE_STEPS = 5
_NUM_ENVS = 2

@register_env("my_env", max_episode_steps=_MAX_EPISODE_STEPS)
class MyEnv(BaseEnv):
    def __init__(self, *args, **kwargs):
        self._first_initialized = False
        super().__init__(*args, **kwargs)

    def _first_initilize_episode(self):
        self._episode_steps = torch.zeros(
            self.num_envs, device=self.device, dtype=torch.int32)

    def _initialize_episode(self, env_idx: torch.Tensor, options: dict):
        if not self._first_initialized:
            self._first_initilize_episode()
            self._first_initialized = True

        self._episode_steps[env_idx] = 0

    def evaluate(self) -> dict:
        self._episode_steps += 1
        fail = self._episode_steps + \
            torch.arange(
                self.num_envs, device=self.device) > _MAX_EPISODE_STEPS  # (num_envs,)
        return dict(fail=fail)

def main():
    env = gym.make(id="my_env",
                   robot_uids='panda',
                   num_envs=2,
                #    render_mode="human",
                   reward_mode="none"
                   )

    episode_steps = torch.zeros(
        _NUM_ENVS, dtype=torch.int32, device='cpu' if _NUM_ENVS == 1 else 'cuda')

    obs, _ = env.reset(seed=0)
    try:
        while True:
            time.sleep(0.1)
            obs, reward, terminated, truncated, info = env.step(None)
            episode_steps += 1
            done: torch = terminated | truncated
            if done.any():
                env_idx = done.nonzero(as_tuple=True)[0]
                print(env_idx.tolist(), episode_steps[env_idx].tolist())
                episode_steps[env_idx] = 0
                options = {
                    'env_idx': env_idx
                }
                obs, _ = env.reset(seed=0, options=options)
            # env.render()

    finally:
        env.close()

if __name__ == '__main__':
    main()

```

StoneT2000, 2024-08-22:
It is truncated earlier because you are checking done = terminated | truncated. truncated has always been correct (it is generated by ManiSkill based on max_episode_steps). However your terminated value turns True earlier sometimes so envs are done at different times and you reset envs based on done, not truncated. Note that terminated is True if the env is success or fail.

Your fail condition (assuming it is meant to imitate a episode timer fail condition) is incorrect. You are doing 
```
fail = self._episode_steps + \
            torch.arange(
                self.num_envs, device=self.device) > _MAX_EPISODE_STEPS  # (num_envs,)
        return dict(fail=fail)
```
self._episode_steps + torch.arange(self.num_envs, ...) > ... is checking time limit wrong.

hesic73, 2024-08-22:
The failure condition isn't related to an episode timer. Instead, it mimics scenarios where parallel environments finish at different timesteps. In my actual environment, the logged episode length sometimes exceeds `max_episode_steps`.

I've identified the issue. `evaluate` is being called not only during `step` but also during `reset`. Therefore, if I need something to be calculated only for each step, I should place it in `_after_control_step`.

closed 2024-08-22 by hesic73

StoneT2000, 2024-08-22:
Ah I see. Yes for anything you want to occur after each step that function would be the one to use.