mani-skill/ManiSkill #522: torque controller feasibility?

Nielsencu, 2024-08-23:
as the tittle suggests, how easy would it be to implement a closed-loop torque controller with ManiSkill?

e.g can we access the force or torque readings on each of the joints?

Nielsencu, 2024-08-31:
Hi @StoneT2000, may I know if you would have some insights on this? Thank you!

StoneT2000, 2024-09-01:
You should be able to access joint force acceleration values so you could penalize it to limit torque. Controllers also have a force limit setting.

There currently may be an issue with setting joint forces however which would be a real torque controller. I'll try and investigate next week. 

For now and from experience doing RL and sim2real, position control (pd joint delta pos) is often sufficient for the majority of use cases.

RoboNuke, 2024-10-10:
@Nielsencu I am working on the same problem and created a fork with a joint torque controller, that works similar to how you interface with actual hardware with saturation limits. [here](https://github.com/RoboNuke/ManiSkill/tree/main)  This may help you, I have some more controllers to implement over the next couple days that may add value for you.