mani-skill/ManiSkill #523: How to improve panda grasp?

dennisushi, 2024-08-23:
Below, I have a video of a motion-planning-based demonstration (please ignore the bad quality, it is what RecordEpisode used by default). The grasp is done using `PandaArmMotionPlanningSolver(...).close_gripper()`, which evidently sets the panda gripper action to -1. As you can see, while the grasp is successful, the object slips. I see the base Panda agent has properties that can be overwritten - for gripper stiffness, damping, force limit, friction -- none of them seem to be affecting the quality of the grasp. Do you have any recommendations?

https://github.com/user-attachments/assets/02a51126-df36-48c6-835f-6e99e63c449a

StoneT2000, 2024-08-23:
It is possible the object might be too heavy, you can try modifying the physical material of the loaded YCB object. The grippers already have a high friction value that should enable it to grasp most things.

It's also possible the grasp is not good, maybe it needs to grab a different part of that bottle.

StoneT2000, 2024-09-23:
@dennisushi has your issue been resolved?

closed 2024-09-30 by StoneT2000