mani-skill/ManiSkill #524: [Question] Simulation Slowdown with Multiple Processes on the Same GPU

hesic73, 2024-08-23:
I've noticed that when I run multiple processes on the same GPU, the simulation time significantly increases. For example, a process that normally takes 4 hours to complete slows down to 8 hours when I run two such processes concurrently on the same GPU. Is this normal?

StoneT2000, 2024-08-24:
Yes. ManiSkill is extremely optimized to the point the GPU is almost always working to its limit when running parallel envs with very little CPU/python overhead. Multiple GPUs are needed if you want to do more parallelization

hesic73, 2024-08-24:
Thank you! And I find that I have trouble installing the latest commit:

```
pip install git+https://github.com/haosulab/ManiSkill.git#egg=mani-skill
Collecting mani-skill
  Cloning https://github.com/haosulab/ManiSkill.git to c:\users\hesicheng\appdata\local\temp\pip-install-3nwnyg7l\mani-skill_6d1f942c27dd4c45b7db695ce6a963f9
  Running command git clone --filter=blob:none --quiet https://github.com/haosulab/ManiSkill.git 'C:\Users\hesicheng\AppData\Local\Temp\pip-install-3nwnyg7l\mani-skill_6d1f942c27dd4c45b7db695ce6a963f9'
  error: invalid path 'docs/source/user_guide/additional_resources/benchmarking_results/rtx_3080/fps:camera_size_rgb.png'
  fatal: unable to checkout working tree
  warning: Clone succeeded, but checkout failed.
  You can inspect what was checked out with 'git status'
  and retry with 'git restore --source=HEAD :/'

  error: subprocess-exited-with-error
  × git clone --filter=blob:none --quiet https://github.com/haosulab/ManiSkill.git 'C:\Users\hesicheng\AppData\Local\Temp\pip-install-3nwnyg7l\mani-skill_6d1f942c27dd4c45b7db695ce6a963f9' did not run successfully.
  │ exit code: 128
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet https://github.com/haosulab/ManiSkill.git 'C:\Users\hesicheng\AppData\Local\Temp\pip-install-3nwnyg7l\mani-skill_6d1f942c27dd4c45b7db695ce6a963f9' did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

closed 2024-08-24 by hesic73

StoneT2000, 2024-08-24:
Interesting regarding the windows issue. But note that GPU simulation only works on linux, windows currently only has cpu sim + rendering working