mani-skill/ManiSkill #525: Issues with using pytorch_kinematics

sean1295, 2024-08-23:
When you do pd_ee_pose control which necessitates the usage of pytorch_kinematics, it throws an error when the xml file contains capsule mesh. Here is the minimal reproducible code. 

`import gymnasium as gym
import mani_skill.envs
import torch
num_envs = 2
env = gym.make("PushT-v1", control_mode='pd_ee_delta_pose', num_envs=num_envs)`

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[5], line 5
      3 import torch
      4 num_envs = 2 # you can go up to 4096 on better GPUs
----> 5 env = gym.make("PushT-v1", control_mode='pd_ee_delta_pose', num_envs=num_envs)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/gymnasium/envs/registration.py:802, in make(id, max_episode_steps, autoreset, apply_api_compatibility, disable_env_checker, **kwargs)
    799     render_mode = None
    801 try:
--> 802     env = env_creator(**env_spec_kwargs)
    803 except TypeError as e:
    804     if (
    805         str(e).find("got an unexpected keyword argument 'render_mode'") >= 0
    806         and apply_human_rendering
    807     ):

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/utils/registration.py:167, in make(env_id, **kwargs)
    165     raise KeyError("Env {} not found in registry".format(env_id))
    166 env_spec = REGISTERED_ENVS[env_id]
--> 167 env = env_spec.make(**kwargs)
    168 return env

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/utils/registration.py:78, in EnvSpec.make(self, **kwargs)
     76         print("Exiting as assets are not found or downloaded")
     77         exit()
---> 78 return self.cls(**_kwargs)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/envs/tasks/tabletop/push_t.py:125, in PushTEnv.__init__(self, robot_uids, robot_init_qpos_noise, *args, **kwargs)
    123 def __init__(self, *args, robot_uids="panda_stick", robot_init_qpos_noise=0.02,**kwargs):
    124     self.robot_init_qpos_noise = robot_init_qpos_noise
--> 125     super().__init__(*args, robot_uids=robot_uids, **kwargs)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/envs/sapien_env.py:318, in BaseEnv.__init__(self, num_envs, obs_mode, reward_mode, control_mode, render_mode, shader_dir, enable_shadow, sensor_configs, human_render_camera_configs, robot_uids, sim_cfg, reconfiguration_freq, sim_backend, render_backend, parallel_in_single_scene)
    314 self._set_main_rng(2022)
    315 self._elapsed_steps = (
    316     torch.zeros(self.num_envs, device=self.device, dtype=torch.int32)
    317 )
--> 318 obs, _ = self.reset(seed=2022, options=dict(reconfigure=True))
    319 self._init_raw_obs = common.to_cpu_tensor(obs)
    320 """the raw observation returned by the env.reset (a cpu torch tensor/dict of tensors). Useful for future observation wrappers to use to auto generate observation spaces"""

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/envs/sapien_env.py:728, in BaseEnv.reset(self, seed, options)
    726     with torch.random.fork_rng():
    727         torch.manual_seed(seed=self._episode_seed)
--> 728         self._reconfigure(options)
    729         self._after_reconfigure(options)
    731 # TODO (stao): Reconfiguration when there is partial reset might not make sense and certainly broken here now.
    732 # Solution to resolve that would be to ensure tasks that do reconfigure more than once are single-env only / cpu sim only
    733 # or disable partial reset features explicitly for tasks that have a reconfiguration frequency

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/envs/sapien_env.py:587, in BaseEnv._reconfigure(self, options)
    585 # load everything into the scene first before initializing anything
    586 self._setup_scene()
--> 587 self._load_agent(options)
    588 self._load_scene(options)
    589 self._load_lighting(options)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/mani_skill/envs/sapien_env.py:370, in BaseEnv._load_agent(self, options)
    366                 raise RuntimeError(
    367                     f"Agent {robot_uid} not found in the dict of registered agents. If the id is not a typo then make sure to apply the @register_agent() decorator."
    368                 )
[... 171 lines cut ...]
File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/pytorch_kinematics/urdf_parser_py/xml_reflection/core.py:517, in Reflection.set_from_xml(self, obj, node, path, info)
    515 else:
    516     if tag in unset_scalars:
--> 517         element.set_from_xml(obj, child, element_path)
    518         unset_scalars.remove(tag)
    519     else:

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/pytorch_kinematics/urdf_parser_py/xml_reflection/core.py:361, in Element.set_from_xml(self, obj, node, path)
    360 def set_from_xml(self, obj, node, path):
--> 361     value = self.value_type.from_xml(node, path)
    362     setattr(obj, self.var, value)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/pytorch_kinematics/urdf_parser_py/urdf.py:139, in GeometricType.from_xml(self, node, path)
    137 children = xml_children(node)
    138 assert len(children) == 1, 'One element only for geometric'
--> 139 return self.factory.from_xml(children[0], path=path)

File ~/anaconda3/envs/maniskill/lib/python3.12/site-packages/pytorch_kinematics/urdf_parser_py/xml_reflection/core.py:251, in FactoryType.from_xml(self, node, path)
    249 cur_type = self.typeMap.get(node.tag)
    250 if cur_type is None:
--> 251     raise Exception("Invalid {} tag: {}".format(self.name, node.tag))
    252 value_type = get_type(cur_type)
    253 return value_type.from_xml(node, path)

Exception: Invalid geometric tag: capsule

StoneT2000, 2024-08-24:
try pip uninstall pytorch_kinematics_ms and then reinstall it

sean1295, 2024-08-24:
Thanks, it works!

closed 2024-08-24 by sean1295