mani-skill/ManiSkill #528: tdmpc2 baseline AttributeError: 'ManiSkillVectorEnv' object has no attribute 'max_episode_steps'

chenyenru, 2024-08-26:
To reproduce:

``` bash
conda env create -f examples/baselines/tdmpc2/environment.yaml  # with sapien dep update
conda activate tdmpc2-ms
python train.py env_id=PushCube-v1
```

Error returned:
```bash
Work dir: /home/chenyenru/Documents/ManiSkill/examples/baselines/tdmpc2/logs/PushCube-v1/1/default
Error executing job with overrides: ['env_id=PushCube-v1']
Traceback (most recent call last):
  File "/home/chenyenru/Documents/ManiSkill/examples/baselines/tdmpc2/train.py", line 55, in train
    env=make_envs(cfg),
  File "/home/chenyenru/Documents/ManiSkill/examples/baselines/tdmpc2/envs/__init__.py", line 31, in make_envs
    cfg.episode_length = env.max_episode_steps
AttributeError: 'ManiSkillVectorEnv' object has no attribute 'max_episode_steps'
```

This is possibly related to the vector env wrapper [`ManiSkillVectorEnv`](https://github.com/haosulab/ManiSkill/blob/4fa913d2112d75952eea4ed525c6e1714b850ea3/mani_skill/vector/wrappers/gymnasium.py), where `max_episode_steps` might not be passed as an attribute.

Should ManiSkillVectorEnv have `max_episode_steps` injected as a property?  

Related gymnasium discussions:
- https://github.com/Farama-Foundation/Gymnasium/issues/615

t-sekai, 2024-08-28:
It's a dependency issue. It was working for mani_skill 3.0.0b5. The code is using the default max_episode_steps for the Maniskill task. However, the newest ManiSkillVectorEnv in 3.0.0b9 doesn't have max_episode_steps as an attribute. 

The workaround would be to pass max_episode_steps in the config to set a max_epsiode_steps, instead of trying to use the default value.

t-sekai, 2024-08-28:
I will update the code on this issue with https://github.com/haosulab/ManiSkill/issues/486.

chenyenru, 2024-08-29:
Thanks! Wondering if the change on `ManiSkillVectorEnv` to not include `max_episode_steps` was intentional?

I see in https://github.com/Farama-Foundation/Gymnasium/issues/615 that it was intentional to not take in the `max_episode_steps` to avoid affecting the `max_episode_steps` within [`TimeLimit`](https://github.com/openai/gym/blob/dcd185843a62953e27c2d54dc8c2d647d604b635/gym/wrappers/time_limit.py) property. 

Will injecting `max_episode_steps` be common when using `ManiSkillVectorEnv`? There seems to be an injection wrapper suggested in https://github.com/Farama-Foundation/Gymnasium/issues/615#issuecomment-1639969801 to find the `max_episode_steps` in `TimeLimit` wrapper and inject that into the base environment.

hesic73, 2024-08-29:
I wrote a custom decorator based on https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/registration.py
```python
import json
from copy import deepcopy
from functools import partial

import gymnasium as gym
from gymnasium.envs.registration import WrapperSpec

from mani_skill import logger

from mani_skill.utils.registration import register, REGISTERED_ENVS, make, make_vec

from typing import List

def register_env(
    uid: str,
    override=False,
    additional_wrappers: List[WrapperSpec] = [],
    **kwargs,
):
    """A decorator to register ManiSkill environments.

    Args:
        uid (str): unique id of the environment.
        override (bool): whether to override the environment if it is already registered.

    Notes:
        - `max_episode_steps` is processed differently from other keyword arguments in gym.
          `gym.make` wraps the env with `gym.wrappers.TimeLimit` to limit the maximum number of steps.
        - `gym.EnvSpec` uses kwargs instead of **kwargs!
    """
    try:
        json.dumps(kwargs)
    except TypeError:
        raise RuntimeError(
            f"You cannot register_env with non json dumpable kwargs, e.g. classes or types. If you really need to do this, it is recommended to create a mapping of string to the unjsonable data and to pass the string in the kwarg and during env creation find the data you need"
        )

    def _register_env(cls):

        max_episode_steps: int = cls.max_episode_steps
        assert max_episode_steps is not None, "max_episode_steps must be set"
        assert isinstance(max_episode_steps,
                          int), "max_episode_steps must be an integer"

        if uid in REGISTERED_ENVS:
            if override:
                from gymnasium.envs.registration import registry

                logger.warn(f"Override registered env {uid}")
                REGISTERED_ENVS.pop(uid)
                registry.pop(uid)
            else:
                logger.warn(
                    f"Env {uid} is already registered. Skip registration.")
                return cls

        # Register for ManiSkill
        register(
            uid,
            cls,
            max_episode_steps=max_episode_steps,
            default_kwargs=deepcopy(kwargs),
        )

        # Register for gym
        gym.register(
            uid,
            entry_point=partial(make, env_id=uid),
            vector_entry_point=partial(make_vec, env_id=uid),
            max_episode_steps=max_episode_steps,
            disable_env_checker=True,  # Temporary solution as we allow empty observation spaces
            kwargs=deepcopy(kwargs),
            additional_wrappers=[
                WrapperSpec(
                    "MSTimeLimit",
                    entry_point="mani_skill.utils.registration:TimeLimitWrapper",
                    kwargs=dict(max_episode_steps=max_episode_steps),
                )
            ]+additional_wrappers,
        )

        return cls

    return _register_env

```

You can now declare an environment like this:

```python
from mani_skill.envs.sapien_env import BaseEnv

@register_env("MyEnv-v0")
class MyEnv(BaseEnv):
    max_episode_steps = 100
```

t-sekai, 2024-08-29:
> Thanks! Wondering if the change on `ManiSkillVectorEnv` to not include `max_episode_steps` was intentional?
> 
> I see in [Farama-Foundation/Gymnasium#615](https://github.com/Farama-Foundation/Gymnasium/issues/615) that it was intentional to not take in the `max_episode_steps` to avoid affecting the `max_episode_steps` within [`TimeLimit`](https://github.com/openai/gym/blob/dcd185843a62953e27c2d54dc8c2d647d604b635/gym/wrappers/time_limit.py) property.
> 
> Will injecting `max_episode_steps` be common when using `ManiSkillVectorEnv`? There seems to be an injection wrapper suggested in [Farama-Foundation/Gymnasium#615 (comment)](https://github.com/Farama-Foundation/Gymnasium/issues/615#issuecomment-1639969801) to find the `max_episode_steps` in `TimeLimit` wrapper and inject that into the base environment.

I made a mistake in my last comment. TD-MPC2 got the default max_episode_steps from gymnasium, not ManiskillVectorEnv. Its likely because the new maniskill package runs on a newer gymnasium version that it can't get max_episode_steps from the environment.

closed 2024-09-30 by StoneT2000