mani-skill/ManiSkill #534: [Feature Request] Enhance `BaseAgent.set_state` to Support Masking Similar to `Actor.set_state`

hesic73, 2024-08-29:
`Actor.set_state` has a parameter `env_idx`that allows partial setting of the state.
https://github.com/haosulab/ManiSkill/blob/edef1ea2199ba5d7d1dcf6275b355c7d2962f7ea/mani_skill/utils/structs/actor.py#L143

In contrast, `BaseAgent.set_state` doesn't accept `env_idx` as input. And `state` should be a dict (but `RecordEpisode` seems to save the flattened array). Would it be possible to extend `BaseAgent.set_state` to support masking, similar to `Actor.set_state`?
https://github.com/haosulab/ManiSkill/blob/edef1ea2199ba5d7d1dcf6275b355c7d2962f7ea/mani_skill/agents/base_agent.py#L318

StoneT2000, 2024-08-29:
Yeah that is some legacy code for robot state recording. We actually don't have any code that uses that. Can look into updating it

hesic73, 2024-08-29:
I find that masking is supported by `Articulation.set_state`:
https://github.com/haosulab/ManiSkill/blob/334fadfd417abcbf1f39c997a1d6715e0167b54f/mani_skill/utils/structs/articulation.py#L258

So I can simply use `BaseAgent.robot.set_state`.

closed 2024-08-29 by hesic73