mani-skill/ManiSkill #537: Missing demos/ Directory in ManiSkill2 v0.5.0

pumfish, 2024-08-30:
I'm working on ManiSkill2 v0.5.0 and have found that my project is missing the ```demos/``` directory. I checked the **README** but did not find any information about this. Additionally, the links under **Demonstrations** are broken.

**Questions:**
- How can I obtain the ```demos/``` directory?
- Is there a step I might have missed, or is there an updated link or source for the demonstrations?
Thank you!

StoneT2000, 2024-08-30:
See https://maniskill.readthedocs.io/en/v0.5.3-rtd/concepts/demonstrations.html, the docs link from before got moved

pumfish, 2024-09-02:
> See https://maniskill.readthedocs.io/en/v0.5.3-rtd/concepts/demonstrations.html, the docs link from before got moved

Thank you very much for your reply. I’ll try it first to see if it works.

closed 2024-09-02 by StoneT2000