mani-skill/ManiSkill #541: Segmentation fault (core dumped) while installing vulkan

brian2lee, 2024-09-03:
Installing maniskill, following the vulkan [instruction](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan) to install vulkan, while `vulkaninfo` showed `Segmentation fault (core dumped)`. Followed the troubleshooting, `/usr/share/vulkan/icd.d/nvidia_icd.json` & `/usr/share/glvnd/egl_vendor.d/10_nvidia.json` exist. Have no idea what's wrong.
Environment:
```
ubuntu 20.04
nvidia-smi 
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.35.03              Driver Version: 560.35.03      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   49C    P0             26W /   80W |      15MiB /   6144MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1318      G   /usr/lib/xorg/Xorg                              4MiB |
+-----------------------------------------------------------------------------------------+

ldconfig -p | grep libGLX_nvidia
	libGLX_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libGLX_nvidia.so.0
```

nvidia_icd.json:
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```
10_nvidia.json
```
{
    "file_format_version" : "1.0.1",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.3.280"
    }
}
```
And just in case here's my pip3 list:
```
Package                       Version             
----------------------------- --------------------
absl-py                       2.1.0               
actionlib                     1.14.0              
angles                        1.9.13              
apturl                        0.5.2               
astunparse                    1.6.3               
attrs                         19.3.0              
autobahn                      17.10.1             
Automat                       0.8.0               
base-local-planner            1.17.3              
bcrypt                        3.1.7               
blinker                       1.4                 
bondpy                        1.8.6               
breezy                        3.0.2               
Brlapi                        0.7.0               
cachetools                    5.5.0               
camera-calibration            1.17.0              
camera-calibration-parsers    1.12.0              
catkin                        0.8.10              
[... 268 lines cut ...]
transforms3d                  0.4.2               
trimesh                       4.4.8               
turtlebot3-teleop             1.2.5               
Twisted                       18.9.0              
txaio                         2.10.0              
typing-extensions             4.5.0               
u-msgpack-python              2.1                 
ubuntu-advantage-tools        8001                
ubuntu-drivers-common         0.0.0               
ufw                           0.36                
unattended-upgrades           0.1                 
urllib3                       1.25.8              
usb-creator                   0.3.7               
vcstools                      0.1.42              
wadllib                       1.3.3               
werkzeug                      3.0.4               
wheel                         0.34.2              
wrapt                         1.11.2              
wsaccel                       0.6.2               
wstool                        0.1.18              
xacro                         1.14.18             
xkit                          0.0.0               
zipp                          1.0.0               
zope.interface                4.7.1  
```

StoneT2000, 2024-09-03:
Can you run vulkaninfo

And then also try reinstalling vulkan.

If vulkaninfo says segmentation fault then there may be something wrong with your drivers, OS, and or vulkan setup. In which case you should post an issue to https://github.com/KhronosGroup/Vulkan-Tools/

brian2lee, 2024-09-03:
I've already tried reinstalling vulkan even the whole maniskill, the outcome is the same. I'll post the issue to https://github.com/KhronosGroup/Vulkan-Tools/ too

StoneT2000, 2024-09-11:
@brian2lee it seems 560 nvidia driver has some issues based on the issue you posted there. I saw you are trying 470, does it work? Normally we expect a much later version (I use 535 atm, 550 should? also work as recommended by the other user)

closed 2024-09-30 by StoneT2000