mani-skill/ManiSkill #557: ImportError: cannot import name '_ccallback_c' from 'scipy._lib'

brian2lee, 2024-09-11:
Running through the [installation](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html) tutorial, when testing installation `python -m mani_skill.examples.demo_random_action` got 
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 102, in _get_module_details
    loader = get_loader(mod_name)
  File "/usr/lib/python2.7/pkgutil.py", line 462, in get_loader
    return find_loader(fullname)
  File "/usr/lib/python2.7/pkgutil.py", line 472, in find_loader
    for importer in iter_importers(fullname):
  File "/usr/lib/python2.7/pkgutil.py", line 428, in iter_importers
    __import__(pkg)
  File "mani_skill/__init__.py", line 22
    def format_path(p: str):
                     ^
SyntaxError: invalid syntax

```
Tried `python3 -m mani_skill.examples.demo_random_action` got
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/brianlee/ManiSkill/mani_skill/__init__.py", line 64, in <module>
    from .envs import *
  File "/home/brianlee/ManiSkill/mani_skill/envs/__init__.py", line 1, in <module>
    from .scenes import *
  File "/home/brianlee/ManiSkill/mani_skill/envs/scenes/__init__.py", line 1, in <module>
    from mani_skill.utils.registration import register_env
  File "/home/brianlee/ManiSkill/mani_skill/utils/registration.py", line 15, in <module>
    from mani_skill.vector.wrappers.gymnasium import ManiSkillVectorEnv
  File "/home/brianlee/ManiSkill/mani_skill/vector/wrappers/gymnasium.py", line 9, in <module>
    from mani_skill.utils.structs.types import Array
  File "/home/brianlee/ManiSkill/mani_skill/utils/structs/__init__.py", line 3, in <module>
    from .base import (
  File "/home/brianlee/ManiSkill/mani_skill/utils/structs/base.py", line 13, in <module>
    from mani_skill.utils.structs.pose import Pose
  File "/home/brianlee/ManiSkill/mani_skill/utils/structs/pose.py", line 10, in <module>
    from mani_skill.utils.geometry.rotation_conversions import (
  File "/home/brianlee/ManiSkill/mani_skill/utils/geometry/__init__.py", line 1, in <module>
    from .geometry import *
  File "/home/brianlee/ManiSkill/mani_skill/utils/geometry/geometry.py", line 7, in <module>
    from scipy.spatial.transform import Rotation
  File "/usr/lib/python3/dist-packages/scipy/__init__.py", line 119, in <module>
    from scipy._lib._ccallback import LowLevelCallable
  File "/usr/lib/python3/dist-packages/scipy/_lib/_ccallback.py", line 1, in <module>
    from . import _ccallback_c
ImportError: cannot import name '_ccallback_c' from 'scipy._lib' (/usr/lib/python3/dist-packages/scipy/_lib/__init__.py)

```
Since when installation requires python >=3.9 (Got the error for python version when `pip install --upgrade git+https://github.com/haosulab/ManiSkill.git`, upgraded python 3.8 to 3.9 and fixed it
), I assumed python 3 is the correct approach. However there's `scipy                         1.3.3` in my `pip3 list`, have no idea how to approach this issue now.

StoneT2000, 2024-09-11:
Try running `pip install scipy==1.14.0`, this version works for me. Alternatively try python3.11 (it is recommended).

StoneT2000, 2024-09-23:
@brian2lee has your issue been resolved?

closed 2024-09-30 by StoneT2000