mani-skill/ManiSkill #563: [Question] replay trajectories in the GPU simulation

YiJiangYue, 2024-09-13:
hi, when i follow the instruction in diffusion policy and  replay trajectories in the GPU simulation:
`python -m mani_skill.trajectory.replay_trajectory \
  --traj-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.h5 \
  --use-first-env-state -c pd_ee_delta_pos -o state \
  --save-traj --num-procs 1 -b gpu --count 100`, 
i got error log:
`raceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/trajectory/replay_trajectory.py", line 337, in <module>
    main(parse_args())
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/trajectory/replay_trajectory.py", line 331, in main
    _main(args)
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/trajectory/replay_trajectory.py", line 259, in _main
    info = action_conversion.from_pd_joint_pos(
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/trajectory/utils/actions/conversion.py", line 186, in from_pd_joint_pos
    return from_pd_joint_pos_to_ee(**locals())
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/trajectory/utils/actions/conversion.py", line 113, in from_pd_joint_pos_to_ee
    pin_model = ori_controller.articulation.create_pinocchio_model()
  File "/mnt/volumes/autopilot-dreamwork-ga/cdq/ManiSkill/mani_skill/utils/structs/articulation.py", line 683, in create_pinocchio_model
    raise NotImplementedError(
NotImplementedError: Cannot create a pinocchio model when GPU is enabled. If you wish to do inverse kinematics you must use fast_kinematics`. 
any solution？

StoneT2000, 2024-09-13:
Can you install the latest git commit of maniskill and try again?

chenyenru, 2024-09-22:
Hi I'm still encountering the error after installing the latest git commit of maniskill via `pip install --upgrade git+https://github.com/haosulab/ManiSkill.git`.

I believe the root cause of this error is that [`create_pinocchio_model()` is called in `from_pd_joint_pos_to_ee`](https://github.com/haosulab/ManiSkill/blob/22a0c8d9be4ad987de58637eccbb701ce169a34b/mani_skill/trajectory/utils/actions/conversion.py#L113) regardless of device type. However, this could be problematic because pinocchio model cannot be created when GPU is enabled, hence why [`NotImplementedError("Cannot create a pinocchio model when GPU is enabled. If you wish to do inverse kinematics you must use fast_kinematics")`](https://github.com/haosulab/ManiSkill/blob/22a0c8d9be4ad987de58637eccbb701ce169a34b/mani_skill/utils/structs/articulation.py#L698C13-L700C14) is raised.

To work around this, I think the initialization of pin_model should be dependent on device type, i.e. use pinocchio model when using cpu and have inverse kinematics when using gpu.

I saw that [`compute_ik`](https://github.com/haosulab/ManiSkill/blob/22a0c8d9be4ad987de58637eccbb701ce169a34b/mani_skill/agents/controllers/utils/kinematics.py#L124) is recently added in `kinematics.py`. For next step, should `kinematics.compute_ik` be used in `from_pd_joint_pos_to_ee` when gpu is used instead of pinocchio model? Thank you

yeknafar, 2024-12-07:
I got this error. any solution ?

ggggfff1, 2024-12-13:
Hi, I'm encountering the same issues. Are there any solutions available?

yeknafar, 2024-12-15:
when I change the control mode to "pd-joint-delta-pose" the problem is solved.

StoneT2000, 2024-12-15:
Currently replaying trajectories and converting action space/controller on the GPU sim is not supported, it is only supported on the CPU. I mark this issue as a request for a feature for now that we will get to in the future

ggggfff1, 2024-12-18:
what is the difference between the trajectory generated on the CPU and GPU?  e.g. trajectory.rgbd.pd_ee_delta_pos.cpu.h5 & trajectory.rgbd.pd_ee_delta_pos.cuda.h5

StoneT2000, 2024-12-19:
The physics engine is slightly different due to different algorithm optimized for cpu / gpu. So this distinction is generally important since training on CPU trajectories and evaluating on GPU sim, especially for harder precise tasks, will do worse than usual (and vice versa)

StoneT2000, 2025-01-30:
should be fixed on the latest main commit, releasing to pypi later.

closed 2025-01-30 by StoneT2000