mani-skill/ManiSkill #566: Installation on windows

j-thapa, 2024-09-17:
I installed maniskill on windows but while trying to run demo example, I got error regarding no insulation of pinocchio; while installing pinocchio it has its own set of errors unfortunately

StoneT2000, 2024-09-17:
for CPU vectorization, the num_envs argument is not to be used as that is for GPU vectorization. Need to update the docs with CPU vectorization examples but there is example located here for RL setups: https://maniskill.readthedocs.io/en/latest/user_guide/reinforcement_learning/setup.html#evaluation

j-thapa, 2024-09-25:
I forgot to mention, I am using Windows and I am trying to train using vector environments in the CPU; is it even feasible to do so. Cause while trying environment using above setup I got multiple errors.

closed 2024-09-26 by j-thapa