mani-skill/ManiSkill #571: [Question] Self-Collision Joints Issue - How to re-model/edit robot mesh structure

CreativeNick, 2024-09-21:
I'm experiencing this bug where the joints/hands of my robot are infinitely spinning when I have self-collisions enabled in [`bimanual_allegro.py`](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/bimanual_allegro.py). 
```
@register_agent()
class Bimanual_Allegro(BaseAgent):
    uid = "Bimanual_Allegro"
    urdf_path = "assets/urdf/ur5e_allegro/robots/dual_ur5e_allegro_inertia_changed.urdf"
    fix_root_link = True
    disable_self_collisions = False # Set to True to disable self-collision
    arm_qpos = np.zeros(12)
    arm_qpos[::2] = left_arm_init_qpos
    arm_qpos[1::2] = right_arm_init_qpos

   ...
```

I'm thinking this is most likely due to the mesh structure of the robot (especially at the joints) clipping/colliding into each other. So when the robot begins moving and certain parts near the joints come into contact, they'll collide and cause a positive feedback loop of the joints spinning uncontrollably in one direction. This severely impacts accuracy and success rate and basically flatlines it to 0.

I'd like to edit and re-model the collision meshes at these areas and was wondering if you could provide some advice/insight on what would be the best way fix this issue? Is it better to edit the collision mesh .STL files ([folder linked here](https://github.com/CreativeNick/SimToReal/tree/553036d72e0289205c0f5566064eed7c162aa032/bimanual-sapien/assets/urdf/ur5e_allegro/meshes/ur5e/collision)) in Blender or edit the [`dual_ur5e_allegro_inertia_changed.urdf`](https://github.com/CreativeNick/SimToReal/blob/553036d72e0289205c0f5566064eed7c162aa032/bimanual-sapien/assets/urdf/ur5e_allegro/robots/dual_ur5e_allegro_inertia_changed.urdf) file? Or is there perhaps another solution that's more straight forward?

Thank you!

Here's a video demonstrating the bug:

https://github.com/user-attachments/assets/3b242536-6a8a-4972-954f-540a01867e06

StoneT2000, 2024-09-21:
First try pip installing the latest git commit of maniskill, there may be some changes that could help.

Otherwise I recommend first running some random actions and opening the viewer and checking which contacts are occurring.

If there are contacts occurring that make no sense (eg 1 finger hitting the robot arm), you can create a .srdg file of the same base name as the urdf and add some rules for which links can't collide. I recommend checking out eg the Fetch robots .srdf file for an example usage

StoneT2000, 2024-09-21:
Also in the viewer i recommend clicking the robot and toggling on collision view to see what the collision mesh looks like

CreativeNick, 2024-09-22:
I pip installed the latest git commit of maniskill (via running `pip install git+https://github.com/haosulab/ManiSkill.git`) but the issue still stands. Here's part of [`view_ycb_env.py`](https://github.com/CreativeNick/SimToReal/blob/553036d72e0289205c0f5566064eed7c162aa032/bimanual-sapien/env_utils/view_ycb_env.py) that runs the random actions:
```
obs, _ = env.reset(seed=0)  # reset with a seed for determinism
done = False
while not done:
    action = env.action_space.sample()
    obs, reward, terminated, truncated, info = env.step(action)
    img = env.render()
```

I'm running `python env_utils/view_ycb_env.py` to view the environments while running random actions.

Here's a video of the collision meshes with self-collisions ENABLED. I'm not too sure how to interpret this since I don't think there are any weird contacts. The joints for the "hands" seem to be the ones that rotate the most, at the very beginning you can see that they rotate almost immediately and stop because they are blocked by the hand colliding with the rest of the arm.

https://github.com/user-attachments/assets/118c8d65-7637-44c6-a173-068232fb8f96

For comparison, here's a video of the collision meshes with self-collisions DISABLED. (The viewport was also a lot smoother which makes sense since it doesn't need to calculate self-collisions):

https://github.com/user-attachments/assets/eaa83aa4-8d42-4637-8532-2b47c37486f6

StoneT2000, 2024-09-24:
Ok hard to tell directly. I will take a look at your codebase and try myself some time this week. Is the main branch good to use to try and test this?

CreativeNick, 2024-09-25:
I'm thinking about getting rid of the the folder of all the YCB assets in the main branch since it has a really large file size and I don't think is necessary. I'll create another branch with those changes in the next 1-2 days and I'll let you know when it's ready. Thank you!

CreativeNick, 2024-09-26:
@StoneT2000 Just updated the repo and tested it on my end, the main branch should be good to install and use. Please let me know if you run into any issues with running it! Thanks once again!

StoneT2000, 2024-10-01:
Something else to try before I try investigating myself deeply:
- Can you try removing every collision tag in your .urdf except for the hand collision tags?
- If the above works, add collision tags back in 1 by 1 until it fails again.
another question:
is the issue particularly about how the hand rotates by itself while sending a 0 action? If so, what controller are you using

CreativeNick, 2024-10-03:
I commented out different collision tags and saved the notable files in this [folder](https://github.com/CreativeNick/SimToReal/tree/main/bimanual-sapien/assets/urdf/ur5e_allegro/robots), to which I ended up narrowing down the wrist **base / palm of the hands** being the **cause of the bug**. I saved a version where I ONLY commented out the left and right wrist bases in this file [`dual_ur5e_allegro_inertia_changed_only_no_palm.urdf`](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/assets/urdf/ur5e_allegro/robots/dual_ur5e_allegro_inertia_changed_only_no_palm.urdf).

You can see in this screenshot that when the palms/wrist base collision tags are commented out, the hands/wrists don't rotating uncontrollably when sending a 0 action (they instead stay still like they're supposed to):
[image]

Yes, the issue is that the hand rotates by itself while sending a 0 action (which is `action = np.zeros_like(env.action_space.sample())` in [`view_ycb_env.py`](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/env_utils/view_ycb_env.py)). 

My control mode is `control_mode="pd_joint_delta_pos"`. The controller configuration can be found in [`bimanual_allegro.py`](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/bimanual_allegro.py):

```
@register_agent()
class Bimanual_Allegro(BaseAgent):
    # ...

@property
    def _controller_configs(self):
        arm_pd_joint_pos = PDJointPosControllerConfig(
            self.arm_joint_names,
            lower=None,
            upper=None,
            stiffness=self.arm_stiffness,
            damping=self.arm_damping,
            force_limit=self.arm_force_limit,
            normalize_action=False,
            friction=self.arm_friction,
        )
        arm_pd_joint_delta_pos = PDJointPosControllerConfig(
            self.arm_joint_names,
            lower=-0.1,
            upper=0.1,
            stiffness=self.arm_stiffness,
            damping=self.arm_damping,
            force_limit=self.arm_force_limit,
            use_delta=True,
            friction=self.arm_friction,
        )
        hand_pd_joint_pos = PDJointPosControllerConfig(
            self.hand_joint_names,
            lower=None,
            upper=None,
            stiffness=self.hand_stiffness,
            damping=self.hand_damping,
            force_limit=self.hand_force_limit,
            normalize_action=True,
            friction=self.hand_friction,
        )

        hand_pd_joint_delta_pos = PDJointPosControllerConfig(
            self.hand_joint_names,
            lower=-0.1,
            upper=0.1,
            stiffness=self.hand_stiffness,
            damping=self.hand_damping,
            force_limit=self.hand_force_limit,
            use_delta=True,
            friction=self.hand_friction,
        )

        controller_configs = dict(
            pd_joint_delta_pos=dict(
                arm=arm_pd_joint_delta_pos, hand=hand_pd_joint_delta_pos
            ),
            pd_joint_pos=dict(arm=arm_pd_joint_pos, hand=hand_pd_joint_pos),
        )
        # Make a deepcopy in case users modify any config
        return deepcopy_dict(controller_configs)
```

StoneT2000, 2024-10-04:
Ok it looks like the issue is clear, you have some self collisions that you want to disable (you don't need to disable all of them).

There are two options here:

1. If you want to keep as many collision meshes as possible (to be close to the real robot)
- You can first check which collision meshes are colliding with which, and then disable those specifically either with code or use a .srdf file (see the fetch robot for how it works). You can check during runtime with the viewer which collisions are occuring and if you notice a pair that shouldn't collide add the pair of links that are colliding to the .srdf file:
[image]
2. Keep your current simplified setup with the bad collisions commented out. This will speed up simulation and is easier to do (it already works for you). Just know that objects can e.g. pass through the palm of the hand now. But sometimes for some tasks this is fine because you might want to only use the e.g. fingers.

CreativeNick, 2024-10-06:
I don't have an .srdf file for my bimanual allegro hand robot. Would it be best to create one for the sake of using the first option to debug this? I don't want to do the second option since I plan on training the model to apply it to a real world robot.

StoneT2000, 2024-10-06:
You have to disable some self collisions regardless unless you fine tune the collision meshes a lot. To avoid disabling all of them the .srdf can define which specific pairs of collisions to ignore.

closed 2024-11-08 by StoneT2000