mani-skill/ManiSkill #572: Questions on installing the vulkan

lozoRUST, 2024-09-23:
Installing the maniskill, here comes the question of unable to locate package vulkan-utils after i run the commande of 'sudo apt install vulkan-utils', and therefore the vulkaninfo also failed.

Besides, i tried to install vulkan througj 'apt install vulkan-tools', and vulkaninfo did work. Despite this, when  'python3 -m mani_skill.examples.demo_random_action' was runned, it informed me of failure to find glnd icd file, also with a runtimerror indicating that deivice "cuda 0" cannot found. 

i am working on the question of cudo0 rightnow, but i'm not sure whether the configuration of vulkan led to failure.

lozoRUST, 2024-09-23:
supplement: i cannot find the glvnd folder in /usr/share/

lozoRUST, 2024-09-23:
CPU infomation:
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         46 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  20
  On-line CPU(s) list:   0-19
Vendor ID:               GenuineIntel
  Model name:            12th Gen Intel(R) Core(TM) i9-12900H
    CPU family:          6
    Model:               154
    Thread(s) per core:  2
    Core(s) per socket:  10
    Socket(s):           1
    Stepping:            3
    BogoMIPS:            5836.80
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss ht syscall n
                         x pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid pni pclmulqdq vmx ssse3 f
                         ma cx16 sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dn
                         owprefetch ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep
                         bmi2 erms invpcid rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves avx_vnni umip waitpkg gfn
                         i vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize flush_l1d arch_capabilities
Virtualization features:
  Virtualization:        VT-x
  Hypervisor vendor:     Microsoft
  Virtualization type:   full
Caches (sum of all):
  L1d:                   480 KiB (10 instances)
  L1i:                   320 KiB (10 instances)
  L2:                    12.5 MiB (10 instances)
  L3:                    24 MiB (1 instance)
Vulnerabilities:
  Gather data sampling:  Not affected
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Mitigation; Enhanced IBRS
  Spec rstack overflow:  Not affected
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl and seccomp
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; Enhanced IBRS, IBPB conditional, RSB filling, PBRSB-eIBRS SW sequence
  Srbds:                 Not affected
  Tsx async abort:       Not affected

lozoRUST, 2024-09-23:
GPU:
[image]

lozoRUST, 2024-09-23:
system information:
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

fbxiang, 2024-09-23:
Are you using ManiSkill's docker? If not you can try to see if that one works and we can start debugging from there. If you are, can you run `sapien info --all` and post the results?

StoneT2000, 2024-09-23:
can you run vulkaninfo and share the results?

Additionally can you run through the troubleshooting here https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#troubleshooting

You may not have installed a few vulkan related packages and are missing the relevant ICD file.

lozoRUST, 2024-09-27:
> 您使用的是 ManiSkill 的 docker 吗？如果没有，您可以尝试看看它是否有效，然后我们可以从那里开始调试。如果是，您可以运行`sapien info --all`并发布结果吗？

Thank for reply! I accutally switch to Ubuntu 22.04 on vmware station right now. Based on the error information, something about cuda have went wrong(i've download and install vulkanSDK instead, i suppose the error is nothing with vulkan itself) . Since I only get the core of linux and have no access to GPU for linux or cloud server, is there any solution for CPU simulation only when setting? Actually i've tried the CPU sim part in colab, it still requires cuda in my computer.

Here's some error information:
[image]

[image]

lozoRUST, 2024-09-29:
Now I have succefully run the maniskil on a GPUserver, ubuntu 20.04. The problem of vulkan is due to the distribution of ubuntu. It seems that ubuntu 22.04 hasn't supported vulkan-utils yet. The trouble shooting solved the other problems succefully.

closed 2024-09-30 by StoneT2000

LukeLIN-web, 2025-04-30:
> Now I have succefully run the maniskil on a GPUserver, ubuntu 20.04. The problem of vulkan is due to the distribution of ubuntu. It seems that ubuntu 22.04 hasn't supported vulkan-utils yet. The trouble shooting solved the other problems succefully.

I wonder whether ubuntu 22.04 can works with maniskill without vulkan-utils? Or it cannot use maniskill at all?