mani-skill/ManiSkill #574: Inconsistent Mesh File Format in panda_v2_gripper.urdf

kh11kim, 2024-09-24:
First of all, thank you for providing such an excellent library! 

While building a custom environment using FloatingPandaGripper, I encountered the following error:

RuntimeError: filesystem error: cannot make canonical path: No such file or directory [franka_description/meshes/visual/finger.dae]

I noticed that the visual meshes are provided in .glb format rather than .dae in the panda_v2_gripper.urdf file. 
After updating the file locally to reflect this, the environment was built successfully.

Could this be updated in the official repository? Thank you!

StoneT2000, 2024-09-24:
Thanks for spotting the bug! It would be great if you could make a PR for this fix, otherwise I can help fix it tomorrow.

StoneT2000, 2024-09-24:
closed by #575

closed 2024-09-24 by StoneT2000