mani-skill/ManiSkill #577: [Question] Best Way to Include Last Actions in Observation

hesic73, 2024-09-24:
What’s the recommended way to add the previous time step’s action to the observation in ManiSkill? Using a `gymnasium.Wrapper`? Or is there a better alternative?

StoneT2000, 2024-09-24:
I would recommend some gym observation wrapper. We have some example wrappers https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/wrappers/flatten.py that demonstrate how to automatically modify observation spaces and generate new observations. For your case you will also need to modify the def step function to save the last action taken

hesic73, 2024-09-24:
Thank you! I've got something like this and it seems to work.

```python
from gymnasium import Wrapper, Env
from gymnasium.spaces import Box
from gymnasium.spaces import Dict as GymSpaceDict

import torch
import numpy as np

from typing import Any, Optional, Tuple, Sequence, Dict, List, Union, Callable

from copy import deepcopy

def get_nested_value(d, key: Union[str, Sequence[str]], dict_types: Sequence[Type] = (dict,)) -> Any:
    if isinstance(key, str):
        return d.get(key)
    current = d
    for k in key:
        if not isinstance(current, dict_types):
            raise KeyError(
                f"Key sequence {key} does not lead to a dictionary. Type: {type(current)}")
        current = current[k]
        if current is None:
            return None
    return current

# Wrapper to add the last action to the tensordict
class LastAction(Wrapper):
    def __init__(self, env: Env, action_dim: int, out_key: str = 'last_action'):
        super().__init__(env)

        self.action_dim = action_dim
        self.out_key = out_key

        self._nested_keys = []  # a nested key to a tensor in the observation dict

        self._update_observation_space()

        self.last_action = None

    def _update_observation_space(self):
        observation_space = deepcopy(self.env.observation_space)
        assert isinstance(
            observation_space, GymSpaceDict), f"Expected observation space to be a GymSpaceDict, got {type(observation_space)}"

        num_envs = None
        dtype = None

        space = observation_space
        while True:
            key = next(iter(space.keys()))
            self._nested_keys.append(key)
            space = space[key]
            if isinstance(space, Box):
                num_envs = space.shape[0]
                sample_tensor = space.sample()
                dtype = sample_tensor.dtype
                break
            assert isinstance(
                space, GymSpaceDict), f"Expected space to be a GymSpaceDict, got {type(space)}"

        assert num_envs is not None
        assert dtype is not None

        # Add the new out_key to the observation space
        observation_space[self.out_key] = Box(
            low=float('-inf'), high=float('inf'), shape=(num_envs, self.action_dim), dtype=dtype
        )

        self._observation_space = observation_space

    def reset(self, *, seed: Optional[int] = None, options: Optional[Dict[str, Any]] = None) -> Dict[str, Any]:
        obs, info = self.env.reset(seed=seed, options=options)
        assert isinstance(
            obs, dict), f"Expected observation to be a dict, got {type(obs)}"

        if self.last_action is None:
            tensor = get_nested_value(obs, self._nested_keys)
            assert isinstance(
                tensor, torch.Tensor), f"Expected tensor to be a torch.Tensor, got {type(tensor)}"
            device = tensor.device
            num_envs = tensor.shape[0]

            self.last_action: torch.Tensor = torch.zeros(
                (num_envs, self.action_dim), device=device)

        if options is None or 'env_idx' not in options:
            self.last_action.zero_()
        else:
            env_idx = options['env_idx']
            self.last_action[env_idx] = 0

        obs.update({self.out_key: self.last_action})
        return obs, info

    def step(self, action: torch.Tensor):
        observation, reward, terminated, truncated, info = self.env.step(
            action)
        assert isinstance(
            observation, dict), f"Expected observation to be a dict, got {type(observation)}"

        assert self.last_action is not None, "Expected last action to be initialized"
        assert self.last_action.shape == action.shape, f"Expected last action shape to be {action.shape}, got {self.last_action.shape}"

        observation.update({self.out_key: self.last_action})
        self.last_action = action

        return observation, reward, terminated, truncated, info

```

closed 2024-09-24 by hesic73