mani-skill/ManiSkill #582: issue while loading checkpoint to evaluate and save video

j-thapa, 2024-09-27:
while trying to load a checkpoint and evaluate on a eval env

python train.py --evaluate --checkpoint='runs/SinkEnv__train__1__1727423959/ckpt_3601.pt' --num_eval_envs=1 --num-eval-steps=100

am getting following error

Traceback (most recent call last):
  File "C:\Users\jthapa\Desktop\repos\coverage-planer\mani_skill\train.py", line 402, in <module>
    print(eval_envs.step(action))
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jthapa\AppData\Local\miniconda3\envs\coverage\Lib\site-packages\mani_skill\vector\wrappers\gymnasium.py", line 89, in step      
    obs, rew, terminations, truncations, infos = self._env.step(actions)
                                                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jthapa\AppData\Local\miniconda3\envs\coverage\Lib\site-packages\mani_skill\utils\wrappers\record.py", line 452, in step
    self._trajectory_buffer.terminated = common.append_dict_array(
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jthapa\AppData\Local\miniconda3\envs\coverage\Lib\site-packages\mani_skill\utils\common.py", line 96, in append_dict_array      
    x2 = x2[:, None, :]
         ~~^^^^^^^^^^^^
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

StoneT2000, 2024-10-01:
What is train.py? Is this the ppo code? If it is the same PPO code from github it may have some issues with using a CPU simulation backend (I haven't modified to code yet to support GPU and CPU sim backends easily)

j-thapa, 2024-10-01:
yes, Its the exact copy of PPO_rgb code file.

closed 2024-10-01 by j-thapa