mani-skill/ManiSkill #588: [Question] static agent contact

j-thapa, 2024-10-01:
I am trying to train a panda arm to polish a surface (sink); the object has been imported as .obj and added as static agent in the task environment; where in case of polisher i just replaced panda gripper hand with a cylindrical body to acts as polisher.

So, my question is I can get contacts but no collision although I can see collision detection option in rendering; also once polishing surface contacts with the sink it behaves awkwardly (which might due to the contact or collision physics of sapien) and doesnt smoothly navigates xy plane of the sink; how and what can I tune in for polisher to behave as a polishing body and glides through the static sink surface while making contact with the surface? 

Does anyone has experience with such application?

StoneT2000, 2024-10-01:
do you have any example videos and or code?

I myself will also release a cleaning task in the near future which you can use as a reference

j-thapa, 2024-10-10:
that would be really great

closed 2025-05-20 by StoneT2000