mani-skill/ManiSkill #602: [Question] How to import mesh with textures on vertices into the Maniskill

TritiumR, 2024-10-03:
Hi. I have one mesh with texture on vertices that I want to import to Maniskill and build actor with it.

I tried this, but it didn't include the texture in the redering. 

builder = self.scene.create_actor_builder()
builder.add_convex_collision_from_file(filename='./normalized.dae')
builder.add_visual_from_file(filename='./normalized.dae')
self.mesh = builder.build_static(name='mesh')

I also tried .obj and .ply. They all contains colors, which could be shown in Meshlab. But when I import them, the color is missing.

Is there any specific way (e.g., different APIs or different format of texture files) to do this?

Thank you.

StoneT2000, 2024-10-04:
.obj should show up. Can you load into eg blender and check there are colors? And try exporting to glb and load it. If it still doesn't work can you share the file here?

StoneT2000, 2024-11-07:
Has your issue been resolved?

TritiumR, 2024-11-08:
Thanks for your message and sorry for the late reply. I baked the texture into the vertex color and it works. Thanks

closed 2024-11-08 by TritiumR

saulgooodman, 2025-03-07:
Hello, I've encountered the same issue. The mesh had colors in MeshLab(I tried .ply and .obj), but the colors disappeared after importing it into Sapien. How can I bake the texture into vertex colors? Thank you!