mani-skill/ManiSkill #608: [Question] How to edit SAPIEN menu UI

CreativeNick, 2024-10-06:
I was wondering if there's a way to edit the UI elements of the SAPIEN menu, such as text/color and even additional functionality)? I understand there's the [SAPIEN](https://github.com/haosulab/SAPIEN) repo, but I'm not sure how to edit it via the ManiSkill repo / my end.
Thanks!

[image]

StoneT2000, 2024-10-08:
It is possible, but we haven't made it easily accessible yet.

I'll add some basic documentation soon about adding plugins / modifying default ones in the viewer

CreativeNick, 2024-12-11:
@StoneT2000 Hey Stone, are there any updates on this? If not no worries!

StoneT2000, 2024-12-16:
Sorry this is a little bit lower priority at the moment.

To help our team though can you detail some examples? Ideally for the documentation we can release a useful tutorial for a somewhat commonly used modification to the viewer

CreativeNick, 2024-12-18:
I was thinking of being able to edit the look and functionality of the UI. For instance, In Godot you can create custom where you can change specific properties like text, colors, fonts, icons, sliders, dropdown menus, etc. A good example is Passivestar's godot-minimal-theme (https://github.com/passivestar/godot-minimal-theme). Even being able to re-arrange certain containers/sections would be something I'm interested in. I'm also interested in being able to create custom plugins or edit the functionality of the UI, like with Blender's plugins. In Blender, you can create plugins that add new buttons, dropdown menus, sliders, etc for the user to use.

StoneT2000, 2025-04-03:
Well the sapien menu / GUI is all python code essentially. You can look at the source code and see how it works for now.