mani-skill/ManiSkill #637: [Bug] plane collisions don't work with contact querying on GPU sim

arth-shukla, 2024-10-19:
## Reproduction:
1. build env (e.g. UnitreeG1Stand-v1)
2. query contacts between feet and ground
3. note that contacts work for sub scene 0, but the rest are zeroed out

StoneT2000, 2024-10-19:
Option 1 to fix is to modify the contact query to make a special condition for the plane collisions (to only check against scene 0s plane collision). The api would save a contact query that uses scene 0's plane collision component 

Generally there is no reason someone would model different plane collisions in parallel environments because unless they're parallel they will collide. So we can reasonably expect scene 0 to have any plane collisions and other scenes don't have them.

StoneT2000, 2024-10-29:
closed by #651

closed 2024-10-29 by StoneT2000