mani-skill/ManiSkill #647: libcuda.so: cannot open shared object file: No such file or directory

vib2810, 2024-10-23:
Hi,
I have been trying to use the Dockerfile given and run the example
```
docker run --rm -it --runtime=nvidia --gpus all maniskill/base python -m mani_skill.examples.benchmarking.gpu_sim
```
I tried both building the Dockerfile locally and pulling the image using `docker pull maniskill/base`

When I run the benchmark, I get the error
```
Traceback (most recent call last):
  File "/opt/conda/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/examples/benchmarking/gpu_sim.py", line 156, in <module>
    main(parse_args())
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/examples/benchmarking/gpu_sim.py", line 40, in main
    env = gym.make(
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 167, in make
    env = env_spec.make(**kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 78, in make
    return self.cls(**_kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/tasks/tabletop/pick_cube.py", line 27, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 229, in __init__
    physx.enable_gpu()
  File "/opt/conda/lib/python3.9/site-packages/sapien/physx/__init__.py", line 45, in enable_gpu
    ctypes.CDLL("libcuda.so", ctypes.RTLD_GLOBAL)
  File "/opt/conda/lib/python3.9/ctypes/__init__.py", line 382, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcuda.so: cannot open shared object file: No such file or directory
```

StoneT2000, 2024-10-23:
Did you setup the nvidia docker container? It's an additional setup after installing docker.

vib2810, 2024-10-30:
Hi, managed to fix it by uninstalling docker and nvidia-docker, and reinstalling both

closed 2024-10-30 by vib2810