mani-skill/ManiSkill #674: To disable rendering during replay on Windows

SkyRainWind, 2024-11-01:
Hello, thanks for your brilliant work! 

After I grabbed raw datasets, I have troubles transfering trajectories to other control mode(say, I would like to use `python -m mani_skill2.trajectory.replay_trajectory --traj-path demos/v0/rigid_body/PickCube-v0/trajectory.h5 --save-traj --target-control-mode pd_ee_delta_pose --obs-mode rgbd` command to obtain a new trajectory called `trajectory.rgbd.pd_ee_delta_pose.h5`).

I use windows connecting to remote server and I disabled x11 forwarding. Is it possible to disable rendering when replaying trajectories?

Compilation information are listed below.
```
$ python -m mani_skill2.trajectory.replay_trajectory --traj-path demos/v0/rigid_body/PickCube-v0/trajectory.h5 --save-traj --target-control-mode pd_ee_delta_pose

0step [00:00, ?step/s][2024-11-01 21:53:44.939] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing

[2024-11-01 21:53:44.939] [svulkan2] [warning] Continue without GLFW.

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/trajectory/replay_trajectory.py", line 530, in <module>
    main(parse_args())
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/trajectory/replay_trajectory.py", line 524, in main
    _main(args)
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/trajectory/replay_trajectory.py", line 350, in _main
    ori_env = gym.make(env_id, **ori_env_kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/shizuru/anaconda3/envs/pcm/lib/python3.11/site-packages/mani_skill2/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Cannot find a suitable rendering device
```

StoneT2000, 2024-11-01:
Is this on docker? Did you check vulkan is installed correctly following our troubleshooting guide: https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#troubleshooting

I also recommend upgrading to ManiSkill 3 instead, we have the same PickCube demonstrations as before.

SkyRainWind, 2024-11-01:
Thanks for your timely reply! I use vscode terminal directly without docker, and the problem remains after I checked `troubleshooting`. The detailed outputs of `vulkaninfo` are listed below.

<details>
<summary>Click to toggle contents of `vulkaninfo`</summary>

```
$ vulkaninfo
'DISPLAY' environment variable not set... skipping surface info
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.131

Instance Extensions: count = 18
====================
        VK_EXT_acquire_xlib_display            : extension revision 1
        VK_EXT_debug_report                    : extension revision 10
        VK_EXT_debug_utils                     : extension revision 1
        VK_EXT_direct_mode_display             : extension revision 1
        VK_EXT_display_surface_counter         : extension revision 1
        VK_KHR_device_group_creation           : extension revision 1
        VK_KHR_display                         : extension revision 23
        VK_KHR_external_fence_capabilities     : extension revision 1
        VK_KHR_external_memory_capabilities    : extension revision 1
        VK_KHR_external_semaphore_capabilities : extension revision 1
        VK_KHR_get_display_properties2         : extension revision 1
        VK_KHR_get_physical_device_properties2 : extension revision 2
        VK_KHR_get_surface_capabilities2       : extension revision 1
        VK_KHR_surface                         : extension revision 25
        VK_KHR_surface_protected_capabilities  : extension revision 1
        VK_KHR_wayland_surface                 : extension revision 6
        VK_KHR_xcb_surface                     : extension revision 6
        VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 3
=======
VK_LAYER_LUNARG_standard_validation (LunarG Standard Validation Layer) Vulkan version 1.0.131, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.2.73, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.1.73, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id  : 0 (llvmpipe (LLVM 12.0.0, 256 bits))
                Layer-Device Extensions: count = 0

Presentable Surfaces:
=====================

error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
Groups:
=======
[... 470 lines cut ...]
---------------------------------------------
        shaderDrawParameters = true

VkPhysicalDeviceTransformFeedbackFeaturesEXT:
---------------------------------------------
        transformFeedback = true
        geometryStreams   = true

VkPhysicalDeviceUniformBufferStandardLayoutFeaturesKHR:
-------------------------------------------------------
        uniformBufferStandardLayout = true

VkPhysicalDeviceVariablePointersFeatures:
-----------------------------------------
        variablePointersStorageBuffer = true
        variablePointers              = false

VkPhysicalDeviceVertexAttributeDivisorFeaturesEXT:
--------------------------------------------------
        vertexAttributeInstanceRateDivisor     = true
        vertexAttributeInstanceRateZeroDivisor = false
```
</details>

I would like to try Maniskill 3. However, the paper i follow currently uses Maniskill 2, so i've kept the compatibility.

StoneT2000, 2024-11-07:
Can you test running `python -m sapien.example.offscreen`

SkyRainWind, 2024-11-08:
Thanks for your reply! I walk through install workflow and troubleshooting again and I refer to some related issues and the results are listed below.

The result of `python -m sapien.example.offscreen`:
<details>
<summary>1</summary>
/home/tyf/anaconda3/envs/pcm/lib/python3.11/site-packages/sapien/_vulkan_tricks.py:37: UserWarning: Failed to find Vulkan ICD file. This is probably due to an incorrect or partial installation of the NVIDIA driver. SAPIEN will attempt to provide an ICD file anyway but it may not work.
  warn(
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/tyf/anaconda3/envs/pcm/lib/python3.11/site-packages/sapien/example/offscreen.py", line 37, in <module>
    main()
  File "/home/tyf/anaconda3/envs/pcm/lib/python3.11/site-packages/sapien/example/offscreen.py", line 9, in main
    scene = sapien.Scene()
            ^^^^^^^^^^^^^^
  File "/home/tyf/anaconda3/envs/pcm/lib/python3.11/site-packages/sapien/wrapper/scene.py", line 25, in __init__
    [sapien.physx.PhysxCpuSystem(), sapien.render.RenderSystem()]
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: invalid PCI string
Segmentation fault (core dumped)
</details>

The results of `sapien info --all`:
<details>
<summary>2</summary>
Traceback (most recent call last):
  File "/home/tyf/anaconda3/envs/pcm/bin/sapien", line 58, in <module>
    info += sapien.render.get_device_summary() + "\n"
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: invalid PCI string
</details>

The results of 
```
>>> import sapien.core as sapien
>>> sapien.SapienRenderer(offscreen_only=True, device="cuda:0")
```
<details>
<summary>3</summary>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tyf/anaconda3/envs/pcm/lib/python3.11/site-packages/sapien/wrapper/renderer.py", line 9, in __init__
    super().__init__()
RuntimeError: invalid PCI string
</details>

Last but not least, I use windows and a remote server to do everything. Maybe it is better for ubuntu?

StoneT2000, 2024-11-08:
You will want to use ubuntu whenever possible. It is the best supported platform.

We have limited support for running sapien/maniskill on windows systems. It can work but when it doesn't, usually it means something is wrong with the nvidia driver installation.

closed 2024-11-28 by StoneT2000