mani-skill/ManiSkill #677: Teleop using VNC connecting cloud server so slow

wzds2015, 2024-11-03:
I am testing teleop example showing on this page: https://maniskill.readthedocs.io/en/latest/user_guide/data_collection/teleoperation.html

I am using RealVNC to connect cloud server. The operation is very slow (show int he video attached). Is there any way to increase the FPS?

Settings,
1. GPU: RTX A6000 50G
2. vulkan: 1.13.204
3. system: ubuntu 22.04
4. Xvfb window: 1024x768x24

StoneT2000, 2024-11-03:
It might be due to a weak GPU (for rendering RTX gpus work best typically). Could you try running python -m mani_skill.examples.demo_random_action -e "StackCube-v1" --render-mode "human" and let me know how smooth that is?

VNC is known (for any simulator as well) to be not very good and have high latency. The best recommendation is to find a machine with a GPU that is attached to a display, or use an app like teamviewer to access a machine with a display.

wzds2015, 2024-11-03:
> python -m mani_skill.examples.demo_random_action -e "StackCube-v1" --render-mode "human"

I tried the command you suggested. Also I lowered the resolution of X11VNC from 1920P to 1024P. Now it looks better. I realized the simulation texture have some artifacts periodically (26s of the video), do you know the reason?

Can you let me know how to turn off ray tracing? Also I might look into TeamViewer as well.

https://github.com/user-attachments/assets/5e6ec2ce-9a27-4599-847a-c970d795c8a6

closed 2024-11-06 by wzds2015