mani-skill/ManiSkill #679: world frame, robot frame and body frame in maniskill3

Lucky-Light-Sun, 2024-11-04:
Hello, I'm new to the robotics domain and have just started exploring ManiSkill3. I find it to be an impressive work! Currently, I'm reading the [user guidance section](https://maniskill.readthedocs.io/en/latest/user_guide/index.html) of the ManiSkill3 documentation.

I am a bit confused about the [Controllers / Action Spaces section](https://maniskill.readthedocs.io/en/latest/user_guide/concepts/controllers.html), especially regarding the **world frame, body frame, and robot frame**. I'm very interested in modifying these variables and observing how they affect the robot's motion.

Could you please help me understand how to modify these parameters? Thank you again for your contributions to the community!

Looking forward to your reply!!

[image]

StoneT2000, 2024-11-04:
world frame is always centered at 0 with identity rotation

root frame relative to world frame is essentially some translation and rotation applied to the world frame. In the image you share you can see the root frame rotates clockwise a little bit and is shifted to the left. Root frame is based on the root pose of the robot/articulation which you can change in a task by doing self.agent.robot.set_pose...

body frame is the frame of the link of the robot the inverse kinematic controller is controlling. For the panda arms we default to using an invisible link placed in between the two grippers (tcp) and the pose of the link being controlled determines the body frame. This can't really be modified and is dependent on the root pose and robot joint angles

Lucky-Light-Sun, 2024-11-09:
The experiment was successful. Thanks for your timely reply!!

closed 2024-11-09 by Lucky-Light-Sun