mani-skill/ManiSkill #684: Question about PDEEPoseController's PD Gains

hesic73, 2024-11-05:
In ManiSkill's `PDEEPoseController`, the `Kp` and `Kd` gains appear to be defined with respect to the joints, rather than for translation and rotation separately, as in some other EE pose controllers. Is there a specific conversion or relationship between these joint-level gains and the EE translation/rotation gains?

https://github.com/haosulab/ManiSkill/blob/61076748d9e2c7254b1863220615c120f5919d7f/mani_skill/agents/controllers/pd_ee_pose.py#L138-L141

StoneT2000, 2024-11-07:
Currently we don't have other kinds of EE controllers with PD specified on translation/rotation. This would involve modifying the inverse kinematics solver essentially which can get complicated.

hesic73, 2024-11-07:
Thank you! I might give this a try, though I'm unsure about the difficulty yet. But since we can calculate the Jacobian, this should be possible?