mani-skill/ManiSkill #685: How to change textures?

AlexandreBrown, 2024-11-06:
Hi,  
I would like to change textures (randomly or via a png file) of the various objects in the scene (eg: before every new episode).  
I managed to change the `base_color` but when I change the textures, nothing happens.  
Any pointers is appreciated.  
The objective is to change textures, camera FOV, lighting and if possible add new objects to evaluate methods for visual generalization.  
```python
render_body = (
            self.base_env.unwrapped.objs["baked_green_cube_3cm"]._objs[0].components[0]
        )

        base_color_texture = RenderTexture2D(
            "/home/user/Documents/SegDAC/Stylized_Metal_Pattern_001_basecolor.png"
        )
        metallic_texture = RenderTexture2D(
            "/home/user/Documents/SegDAC/Stylized_Metal_Pattern_001_metallic.png"
        )
        normal_texture = RenderTexture2D(
            "/home/user/Documents/SegDAC/Stylized_Metal_Pattern_001_normal.png"
        )
        roughness_texture = RenderTexture2D(
            "/home/user/Documents/SegDAC/Stylized_Metal_Pattern_001_roughness.png"
        )

        for shape in render_body.render_shapes:
            for part in shape.parts:

                material = part.material
                material.base_color = [
                    random.random(),
                    random.random(),
                    random.random(),
                    1.0,
                ]
                # material.base_color_texture = base_color_texture
                # material.normal_texture = normal_texture
                # material.metallic_texture = metallic_texture
                # material.roughness_texture = roughness_texture

        obs_dict, _ = self.base_env.reset()
```

StoneT2000, 2024-11-07:
Is this your own custom environment? What environment is this exactly? And are you planning to use the GPU sim + rendering?

AlexandreBrown, 2024-11-07:
Hi @StoneT2000 , I am using SimplerEnv and TorchRL.  
The code is a TorchRL Env that wraps SimplerEnv environment to utilize it in TorchRL unified interface.  
TorchRL env wrapper:  
```python
import torch
import numpy as np
from tensordict import TensorDict, TensorDictBase
from torchrl.envs import EnvBase
from torchrl.data import Composite, Unbounded, Bounded
from sapien.pysapien.render import RenderTexture2D
import sapien

class SimplerEnvWrapper(EnvBase):
    def __init__(self, base_env, **kwargs):
        super().__init__(**kwargs)
        self._device = torch.device(kwargs.get("device", "cpu"))
        self.base_env = base_env
        self.numpy_to_torch_dtype_dict = {
            bool: torch.bool,
            np.uint8: torch.uint8,
            np.int8: torch.int8,
            np.int16: torch.int16,
            np.int32: torch.int32,
            np.int64: torch.int64,
            np.float16: torch.float16,
            np.float32: torch.float32,
            np.float64: torch.float64,
        }
        self._make_specs()

    def _make_specs(self):
        raw_observation_spec = self.get_image_from_maniskill3_obs_dict(
            self.base_env, self.base_env.observation_space.spaces
        )
        height = raw_observation_spec.shape[-3]
        width = raw_observation_spec.shape[-2]
        self.channels = raw_observation_spec.shape[-1]
        shape = (height, width, self.channels)
        observation_spec = {
            "pixels": Bounded(
                low=torch.from_numpy(
                    raw_observation_spec.low[0, :, :, : self.channels]
                ).to(self._device),
                high=torch.from_numpy(
                    raw_observation_spec.high[0, :, :, : self.channels]
                ).to(self._device),
                shape=shape,
                dtype=torch.uint8,
                device=self._device,
            )
        }
        self.observation_spec = Composite(**observation_spec)

        action_space = self.base_env.action_space
        self.action_spec = Bounded(
            low=torch.from_numpy(action_space.low).to(self._device),
            high=torch.from_numpy(action_space.high).to(self._device),
            shape=action_space.shape,
            dtype=self.numpy_to_torch_dtype_dict[action_space.dtype.type],
            device=self._device,
        )

        self.reward_spec = Unbounded(
            shape=(1,), dtype=torch.float32, device=self._device
        )
        self.done_spec = Unbounded(shape=(1,), dtype=torch.bool, device=self._device)

    def get_image_from_maniskill3_obs_dict(self, env, obs, camera_name=None):
        if camera_name is None:
            if "google_robot" in env.unwrapped.robot_uids.uid:
[... 91 lines cut ...]
    render_mode="rgb_array",
    sim_backend=cfg["env"]["device"],
)
```  
I am testing the following **existing** environments from maniskill3 (using SimplerEnv):  
- `PutCarrotOnPlateInScene-v1`
- `PutCarrotOnPlateInScene-v1`
- `PutSpoonOnTableClothInScene-v1`
- `StackGreenCubeOnYellowCubeBakedTexInScene-v1`
(Since currently only these are supported by SimplerEnv/Maniskill3 integration).  

My goal is to leverage the flexibility of maniskill3/simplerenv and be able to :  
- Randomize colors
- Randomize textures (eg: Assign random textures from the textures in the scene)
- Assign specific textures (I want to come up with specific textures that are challenging and never seen during training for a visual RL generalization benchmark)
- Change camera FOV & positions (again for visual generalization evaluation purposes)
- Change lighting conditions
- Load distracting objects or be able to load a scene with distractions like we could in Maniskill2 (eg: random extra objects in the scene) 
- Load rgb overlay / video overlay 

The more I can achieve from this list, the better. I also would like the simplest changes possible (creating an environment from scratch is not something I am used to in Maniskill3).  
Ideally I would like to apply these randomization at the start of the episode but if video overlay is easy then I guess this specific case requires per step update (if we treat a video as a sequence of frames).  
I understand that GPU vectorization probably means these use cases are much harder, in which case I would prefer to go for the low hanging fruit first (eg: randomization that are only applied at the start of the episode, if that's easier).

Yes I plan on using the GPU to improve simulation performance (fps), I assume that `sim_backend='cuda'` is what needs to be done for this but please feel free to tell me more about it. GPU vectorization is a strong motivation for me to use maniskill3 with simplerenv (via their [maniskill3 branch](https://github.com/simpler-env/SimplerEnv/tree/maniskill3)) instead of the existing maniskill2/simplerenv.

StoneT2000, 2024-11-08:
Thanks for the extensive notes, all of what you suggest are possible but it depends a little bit on what models you want to evaluate actually.

- In particular do you plan to train a model and evaluate it? Or evaluate off the shelf models?
- How realistic do you want the environment to look? Are you planning to try vision based sim2real or just do real2sim evaluation of a model trained on real world data?

There are two ways forward. The easiest option actually is to build a new table-top environment (take one of the templates or e.g. the pick cube environment) and add the parallelizations / randomizations you want for a custom environment. Only choose this option if you don't need to verify real2sim alignment and just simply want a controllable robot and objects.

Alternatively you can copy the code for the bridge dataset digital twins and modify the attributes in there to change the default RGB overlays, swap the overlay at each timestep when using video, modify the scene loader to add distractor objects etc.

https://github.com/haosulab/ManiSkill/blob/56dcd4cf1b1f04b7e7dfd82ec625c8428ce1f801/mani_skill/envs/tasks/digital_twins/bridge_dataset_eval (copy both).

Let me know which option you think is needed and I can suggest the relevant docs/code to do what you want.

AlexandreBrown, 2024-11-11:
Thanks a lot @StoneT2000 for the amazing reply!  
> do you plan to train a model and evaluate it? Or evaluate off the shelf models?   

I plan on training and evaluating models (training from scatch).   

> How realistic do you want the environment to look? Are you planning to try vision based sim2real or just do real2sim evaluation of a model trained on real world data?   

I want to train in simulation using an environment that is as realistic as possible (visually) but if this hinders training time I'm open to try to train using a hybrid approach where the environments are still realistic but maybe slightly less (eg: without ray-tracing) to boost collection speed during training and then the visual generalization benchmark can be more realistic and slower.  
Basically I will need to train agents from scratch in simulation and then once trained, I will evaluate the approach using aggressive visual domain randomization (aggressive sim2real visual changes like random camera FOV, random objects colors, random textures, random lighting etc). The model will only depend on image observation (RGB pixels) and will be trained in an online RL fashion.  

>  The easiest option actually is to build a new table-top environment (take one of the templates or e.g. the pick cube environment) and add the parallelizations / randomizations you want for a custom environment.      

This sounds interesting as I also want not just 1 environment but at least 2-3 that can show increasing level of difficulty (eg: easy to hard).  
Is it easier to create an environment from scatch or to start from an existing one ? Context : I have very little experience in environment design. Where can I find a template and documentation for this ? Where and how can I add the randomization ? Which functions do I have to override and when should what functions be called etc ?   
When you say "add the parallelizations" what do you mean exactly?

AlexandreBrown, 2024-11-18:
@StoneT2000 After looking at the doc for Maniskill3, I'm tempted to use Maniskill3 directly instead of SimplerEnv. Would it be feasible to use Maniskill3 directly while also be able to add the visual distractions ? 

Any help is appreciated!

StoneT2000, 2024-11-23:
Yes using ManiSkill3 directly is likely better. SIMPLER's setup is based on maniskill2 and not well parallelized whereas maniskill3 is.

What kind of visual distractions were you thinking? Like spawning irrelevant objects in a table? (that is possible, just follow the custom tasks tutorial on loading objects and initializing them).

As for more parallel domain randomizations https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html details some of them.

I will mark this issue as a request to add more docs (eg randomizing textures, robot controllers PD parameters and more).

AlexandreBrown, 2024-11-23:
Thanks for the detailed answer @StoneT2000 , I will use ManiSkill3 directly then and I will focus on the simple/default rendering quality which is already fairly good anyways.  
  
> What kind of visual distractions were you thinking?  

I would like the following visual randomization :
- Randomize/Set colors of the objects in the scene (eg: change a table to a specific or random RGB color, change the walls etc). 
- Randomize/Set textures (eg: Assign random textures from the textures in the scene or from files on disk).
  - I want to come up with specific textures that are challenging and never seen during training for a visual RL generalization benchmark.
- Change camera FOV & positions (again for visual generalization evaluation purposes)
- Change lighting conditions (randomize/set variables that affect the lighting)
- Load distracting objects or be able to load a scene with distractions like we could in Maniskill2 (eg: Load extra objects in the scene)     
  - Ideas: Load objects from same scene but as extra object (eg: extra cube), Load a random object from an other scene/env, Load a custom object from a file on disk. 
- Load rgb overlay / video overlay, from a video on disk, change the walls or table or objects to use as texture a frame from a video.  

> I will mark this issue as a request to add more docs  

Makes sense, these might already be possible and all is needed is some documentation improvement.

StoneT2000, 2024-11-27:
Hi @AlexandreBrown I have updated the documentation to include as many details about these here: https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html

StoneT2000, 2024-11-27:
Also for distraction objects, that is simple to do, in your custom task just load more objects and place them on e.g. a table. See https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/index.html for a tutorial on building your own tasks.

AlexandreBrown, 2024-11-28:
Wow @StoneT2000 , thank you so much! I will go through this and report back on my experience. 

Regarding the textures, is it possible to load one from disk ? If so how would we do that with Maniskill3? What format would the file need to be ?

Thanks!

StoneT2000, 2024-11-29:
Any .png / .jpg file can work. 
```
mat = sapien.render.RenderMaterial()
mat.base_color_texture = sapien.render.RenderTexture2D(path_to_file)
builder.add_box_visual(..., material=mat)
```
something like the above is how you would create a box object with a custom color texture to display.

https://learnopengl.com/Getting-started/Textures this might help if you want to understand how 2D textures work more in depth, and how we map 2D images to 3D triangle meshes. We may have some better documentation/guides on rendering features in the future.