mani-skill/ManiSkill #686: Weird texture for ArchitecTHOR_SceneManipulation-v1 Env

wzds2015, 2024-11-06:
I am testing demo_random_action using ArchitecTHOR_SceneManipulation-v1 env. The texture doesn't look good. Anyone has idea what I did wrong?

https://github.com/user-attachments/assets/af678cfb-96df-42a6-9b1c-4a90845e2df1

Also seems teleop doesn't support ArchitecTHOR_SceneManipulation-v1. What modification should I do to add a new env into teleop?

StoneT2000, 2024-11-07:
Currently we don't have easy to use teleop tools yet for mobile manipulation. It is something we will add in the future. It generally requires some engineering effort to essentially keyboard key presses to various actions.

As for the rendering, the AI2 THOR scenes do not have backfaces so you have to move inside the house to see inside it more clearly. If you click an object it will add transparency to it to show you selected it. 

If you want a higher rendering quality you can turn on ray-tracing. Note that only ReplicaCAD and Robocasa scenes have tuned lighting. The AI2THOR scenes don't have lighting defined so it might not look that good unless you add it.

closed 2024-11-28 by StoneT2000

Ethan-Chen-plus, 2025-05-22:
Hi! I have added teleop: https://github.com/haosulab/ManiSkill/pull/1076
Hopes working well!

Ethan-Chen-plus, 2025-05-22:
And the same, I have seen that ai2thor has no ray-tracing since I use `rt-fast` and I will get nothing since the terminal will tell me that all the scene was transparent. For better display quality, I think this function is necessary. 🤔