mani-skill/ManiSkill #693: Add new custom robot failed

ruoxianglee, 2024-11-11:
Hi,

I am trying to add a new robot into Maniskill environment. The robot code is [here](https://github.com/ruoxianglee/ManiSkill/blob/delta-robot/mani_skill/agents/robots/drv70l/drv70l.py).

Following the [tutorial](https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/intro.html, the demo_robot), demo_robot  is invoked as [here](https://github.com/ruoxianglee/ManiSkill/blob/delta-robot/mani_skill/agents/robots/drv70l/test.py).

At first, it works properly. However, after pulling the new changes of Maniskill, I encounter the following issue. It seems related to this [update](https://github.com/haosulab/ManiSkill/pull/655), but I am not sure the reason for this.

> 2024-11-11 17:04:12,744 - mani_skill  - WARNING - Agent drv70l_robotiq is already registered. Skip registration.
Traceback (most recent call last):
  File "/home/eric/anaconda3/envs/maniskill/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/eric/ManiSkill/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/home/eric/ManiSkill/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/home/eric/ManiSkill/mani_skill/envs/tasks/empty_env.py", line 25, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/home/eric/ManiSkill/mani_skill/envs/sapien_env.py", line 335, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/home/eric/ManiSkill/mani_skill/envs/sapien_env.py", line 832, in reset
    self._reconfigure(options)
  File "/home/eric/ManiSkill/mani_skill/envs/sapien_env.py", line 670, in _reconfigure
    self._load_agent(options)
  File "/home/eric/ManiSkill/mani_skill/envs/tasks/empty_env.py", line 38, in _load_agent
    super()._load_agent(options, sapien.Pose())
  File "/home/eric/ManiSkill/mani_skill/envs/sapien_env.py", line 419, in _load_agent
    agent: BaseAgent = agent_cls(
TypeError: __init__() got an unexpected keyword argument 'initial_pose'

> During handling of the above exception, another exception occurred:

> Traceback (most recent call last):
  File "/home/eric/ManiSkill/mani_skill/agents/robots/drv70l/test.py", line 3, in <module>
    demo_robot_script.main()
  File "/home/eric/ManiSkill/mani_skill/examples/demo_robot.py", line 35, in main
    env = gym.make(
  File "/home/eric/anaconda3/envs/maniskill/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 814, in make
    raise type(e)(
TypeError: __init__() got an unexpected keyword argument 'initial_pose' was raised from the environment creator for Empty-v1 with kwargs ({'obs_mode': 'none', 'reward_mode': 'none', 'enable_shadow': True, 'control_mode': 'pd_joint_pos', 'robot_uids': 'drv70l_robotiq', 'sensor_configs': {'shader_pack': 'default'}, 'human_render_camera_configs': {'shader_pack': 'default'}, 'viewer_camera_configs': {'shader_pack': 'default'}, 'render_mode': 'human', 'sim_config': {'sim_freq': 100, 'control_freq': 20}, 'sim_backend': 'auto'})

Here is the environment:

> gymnasium                 0.29.1                 
mani-skill                3.0.0b12                 
pip                       24.2      
python                    3.9.20               
pytorch                   2.3.0          
pytorch-kinematics        0.7.4                  
pytorch-seed              0.2.0              
sapien                    3.0.0b1                
scipy                     1.13.1                
setuptools                75.1.0      
torch                     2.5.1                
torchaudio                2.5.1                    
torchtriton               2.3.0             
torchvision               0.20.1

Thanks.

StoneT2000, 2024-11-11:
The bug is here

https://github.com/ruoxianglee/ManiSkill/blob/305da944f12be79ff0a9034508873ac23e8de831/mani_skill/agents/robots/drv70l/drv70l.py#L45-L73

I recommend following the current panda.py arm implementation, you don't need to override the original `__init__` function. You can, but make sure to pass any *args, **kwargs from __init__ to the super __init__.

ruoxianglee, 2024-11-13:
Thanks. It works.

closed 2024-11-13 by ruoxianglee