mani-skill/ManiSkill #694: Initialize the robot arm posture in a custom environment

LuckyCZD, 2024-11-12:
I followed the tutorial to load my custom robotic arm (panda's end effector was replaced with a dexterous hand) into the custom environment. The results I got by setting the keyframe 'rest' and running test.py were normal(picture blow), but when I loaded the robotic arm into a custom environment, the initial position of the robotic arm was all joints = 0. How can I set the initial position of the robot arm in the custom environment to the keyframe position?
[image]

StoneT2000, 2024-11-13:
In your custom environment you should define a `def _initialize_episode` function which upon env.reset will initialize all objects accordingly including the robot. In this function you can do something like `self.agent.robot.set_qpos(self.agent.keyframes["rest"])`.

LuckyCZD, 2024-11-13:
Thank you very much, it's worked.

closed 2024-11-13 by LuckyCZD