mani-skill/ManiSkill #695: Feature Request: Add `conda` installation support for ManiSkill

songyuc, 2024-11-13:
Hi ManiSkill Developers,

Thank you for your amazing work on ManiSkill! This tool has been incredibly helpful for robotics simulation and training.

I am a beginner in Python and use `conda` environments frequently for managing dependencies and isolating my projects. While the current `pip`-based installation method works well, I believe adding support for `conda` installation could provide significant benefits, including:

1. **Beginner-friendliness**: `conda` simplifies dependency management and avoids many common installation issues, especially for users less familiar with Python.
2. **Improved compatibility**: Precompiled packages in `conda` reduce the chances of dependency conflicts or system compatibility issues.
3. **Wider adoption**: Many researchers and students in the robotics and machine learning community rely on `conda` for managing environments.

Would it be possible to consider adding a `conda` installation method for ManiSkill in the future? For example, providing a Conda Forge recipe or maintaining a channel with the necessary builds.

Thank you for considering this suggestion, and I look forward to ManiSkill's future development!

Best regards,  
Yucheng Song

fbxiang, 2024-11-19:
SAPIEN and ManiSkill should be fully compatible with conda if you use the pip provided by conda. We carefully engineered the packages to not rely on system packages. And pip is the only choice for people who do not or cannot use conda.

If you find yourself compiling SAPIEN from source, could you provide more insights on what you are trying to achieve? If you are trying to get SAPIEN to work on Windows or Python3.13, you can get the nightly build here https://github.com/haosulab/SAPIEN/releases (there is practically no difference from the pypi released version).
This is actually built by GitHub CI, and as long as you follow the build steps in CI, you can also build it yourself. If your goal is to build it for a non x86 platform, you can share your experience. If you are targeting MacOS, please note that SAPIEN renderer will not be compatible with the latest M series since many Vulkan features are currently missing for the GPUs, so I really do not recommend building on MacOS.

traversaro, 2024-11-19:
> SAPIEN and ManiSkill should be fully compatible with conda if you use the pip provided by conda. We carefully engineered the packages to not rely on system packages. And pip is the only choice for people who do not or cannot use conda.

Thanks a lot for that! I know from direct experience that this is not straightforward, so thanks a lot for that. I think that is indeed is that the OP was referencing with "pip-based installation method works well". And just to be super-clear, a conda package for manifskill/sapien would not be a replacement or something that can be replacing the `pip` package (my bad if I gave that impression).

> If you find yourself compiling SAPIEN from source, could you provide more insights on what you are trying to achieve?

This is just speaking for me/the lab were I work, I definitely not speaking for the OP. When we add dependency for a mixed C++/Python package, we typically prefer to depend on it via conda-forge rather then pypi for several reasons, the main is that we typically use a lot of C++ dependencies (with a depth of the dependency tree that can reach ~5/10), along Python dependencies. Having all our dependencies on conda-forge permits us to upload to conda-forge any new open source package that we develop in a simple a straightforward way. As soon as we add a dependency that is not on conda-forge, we can't package the resulting downstream package on conda-forge, and typically if the downstream package also have a non-trivial C++ dependency chain (either directly or indirectly), we can't easily package it on PyPI either.

Just to clarify, I do not want to convince you to work on preparing/maintaining a conda-forge package, just explaining why I looked in this in the past (and I did not went forward on working on it, at least for now : ) ). I just dumped some details as I could not find these details anywhere, and they could be interesting if anyone was also looking in making a conda/conda-forge package.

closed 2025-06-27 by songyuc