mani-skill/ManiSkill #696: AttributeError: 'Viewer' object has no attribute 'cpu'

fengjungui, 2024-11-14:
When I use demo_manual_control.py,I meet this problem:
  **render_frame = env.render().cpu().numpy()[0]

  if after_reset:
      after_reset = False
      # Re-focus on opencv viewer
      if args.enable_sapien_viewer:
          renderer.close()
          renderer = visualization.ImageRenderer()
          pass
  # -------------------------------------------------------------------------- #
  # Interaction
  # -------------------------------------------------------------------------- #
  # Input
  renderer(render_frame)**

StoneT2000, 2024-11-14:
What script did you run on the command line?

fengjungui, 2024-11-15:
demo manual control

 

 
&mdash;

StoneT2000, 2024-11-15:
I can't understand what you sent?

closed 2024-11-28 by StoneT2000