mani-skill/ManiSkill #707: Deprecation warnings related to `env.max_episode_steps` in ManiSkill examples

songyuc, 2024-11-20:
Hi,
I am currently working with the ManiSkill library and running the PPO example script. While executing the script, I encountered several deprecation warnings related to accessing `env.max_episode_steps`. Below is the warning message I received:
UserWarning: WARN: env.max_episode_steps to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.max_episode_steps` for environment variables or `env.get_wrapper_attr('max_episode_steps')` that will search the reminding wrappers.

### **Steps to Reproduce:**

1. **Environment Setup:**
   - I am using a conda environment with Python 3.11.
   - The environment is activated using `conda activate maniskill_env`.

2. **Command Executed:**
   ```bash
   python ppo.py --env_id="PushCube-v1" --num_envs=2048 --update_epochs=8 --num_minibatches=32 --total_timesteps=2_000_000 --eval_freq=10 --num-steps=20
   ```

3. **Observed Behavior:**
   - The script runs and starts the training process.
   - During execution, multiple deprecation warnings are logged, indicating that `env.max_episode_steps` is deprecated and will be removed in version 1.0.

### **Example Warning Message:**
```
/home/songyuc/anaconda3/envs/maniskill_env/lib/python3.11/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.max_episode_steps to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.max_episode_steps` for environment variables or `env.get_wrapper_attr('max_episode_steps')` that will search the reminding wrappers.
  logger.warn(
```

### **Expected Behavior:**
- The script should execute without any deprecation warnings.
- It would be great if the ManiSkill developers could provide guidance or updates to the documentation on how to properly access `max_episode_steps` in the current and future versions.

### **Additional Information:**
- **Environment Details:**
  - ManiSkill version:  3.0.0b12
  - Python version: 3.11
  - Operating System: Ubuntu 22.04

- **Related Dependencies:**
  - I have installed all necessary dependencies as per the ManiSkill documentation.

StoneT2000, 2024-11-20:
This is a to-be-deprecated feature from gymnasium although we currently do not plan to upgrade to gymnasium 1.0 soon (you may however use gymnasium 1.0 backwards compatability wrappers)

I'll take a look to see where we are calling this though.

closed 2024-11-20 by StoneT2000

songyuc, 2024-11-21:
Hello @StoneT2000, 
I want to express my gratitude for your team's contributions to ManiSkill. You mentioned that this feature will be deprecated in Gymnasium 1.0, but ManiSkill doesn't plan to upgrade to Gymnasium 1.0 soon.

Given this information, I would like to ask what is the "most recommended" version of Gymnasium to use with ManiSkill currently? Is there a specific version that is most compatible with the current stable version of ManiSkill?

Furthermore, if possible, do you have any plans to update ManiSkill in the future to support Gymnasium 1.0, or can you share any roadmap regarding this matter?

Thank you very much for your help and your response!

StoneT2000, 2024-11-21:
The recommended version is the one that comes installed with ManiSkill (it is pinned to 0.29 in the setup.py file)

As for whether we will natively support gymnasium 1.0 remains to be seen. I agree with the points raised by the TorchRL people about why we don't want to upgrade now: https://github.com/pytorch/rl/discussions/2483. Tldr; too many breaking changes that are silent and will need refactoring of a lot of people's RL code.

If you want to use gymnasium 1.0 ManiSkill might be compatible for the most part provided you use one of their backwards compatibility wrappers for older versions of gymnasium.