mani-skill/ManiSkill #716: CUDA Out of Memory Error While Training with PPO on ManiSkill

songyuc, 2024-11-24:
Dear ManiSkill PPO Developers,

I hope you are doing well. I am currently working on training a PPO agent using the ManiSkill environment and have encountered a CUDA Out of Memory (OOM) error. I would appreciate any advice or suggestions you have about fixing this problem.

### **Error Details**

I am receiving the following error when running the training script:

```bash
Epoch: 102, global_step=1292800
SPS: 2081
Epoch: 103, global_step=1305600
Traceback (most recent call last):
  ...
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 188.00 MiB. GPU 0 has a total capacity of 10.75 GiB of which 177.69 MiB is free. Including non-PyTorch memory, this process has 10.56 GiB memory in use. Of the allocated memory 3.88 GiB is allocated by PyTorch, and 468.36 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation. See documentation for Memory Management.
```

### **Steps to Reproduce**

1. **Environment Setup**: 
   - Operating System: Ubuntu 22.04
   - GPU: NVIDIA RTX 2080 with 11GB memory
   - Python Version: 3.11
   - ManiSkill and PPO code cloned from the latest main branch.

2. **Command Used**:
   ```bash
   python ppo_rgb.py --env_id="PickCube-v1" --num_envs=256 --update_epochs=8 --num_minibatches=8 --total_timesteps=10_000_000
   ```

### **Request for Assistance**
What are the recommendations or best practices for running PPO on ManiSkill with limited GPU memory?

Thank you very much for your time and assistance. I am looking forward to your suggestions and hope to resolve this issue soon.

Best regards,
Yucheng Song

StoneT2000, 2024-11-24:
With less gpu memory available you need to tune ppo down, use less parallel environments or take less steps per parallel environment in order to reduce the replay buffer size. Most of the baseline scripts are tuned assuming 15GB of memory is available.

songyuc, 2024-11-25:
Thanks for @StoneT2000 your response! As a RL beginner, I'd like to confirm my understanding:

1. To resolve the CUDA OOM error, I should adjust these parameters for `python ppo_rgb.py --env_id="PickCube-v1" \
  --num_envs=256 --update_epochs=8 --num_minibatches=8 \
  --total_timesteps=10_000_000` as:
   - Reduce `num_envs` (currently 256)
   - Reduce `num-steps` (currently default)
   - Adjust `num_minibatches` (currently default)

2. Could you suggest reasonable values for these parameters with an RTX 2080 (11GB memory)?

Thank you for your patience and guidance!

songyuc, 2024-11-29:
Thank you for your suggestion on reducing the `--num_envs` parameter to 128 for my CUDA out of memory issue. I appreciate your understanding and help in navigating this problem.

Regarding the num_envs setting, I noticed that in the baseline script provided on the [ManiSkill PPO examples page](https://github.com/haosulab/ManiSkill/tree/main/examples/baselines/ppo), the --num_envs is set to 256 for the PickCube-v1 environment. However, considering the GPU memory limitations I'm experiencing with my NVIDIA RTX 2080 (11GB memory), I understand that reducing this number can help alleviate the memory pressure.

I also observed that the num_envs values provided in the examples are multiples of 8. Would it be advisable to stick to multiples of 8 when adjusting this parameter, or is it flexible based on the available GPU memory and specific training needs?

Thank you once again for your guidance and support.

Best regards,
Yucheng Song

StoneT2000, 2024-11-29:
Best to stick to multiples of 2. The underlying GPU sim most likely will allocate GPU resources more efficient when it is multiples of 2 due to how CUDA works.

songyuc, 2024-11-29:
Thanks sincerely for your guide!

closed 2024-11-29 by songyuc