mani-skill/ManiSkill #722: [bug] Unable to Run train.py in Diffusion Policy Example Code

Rui-li023, 2024-11-28:
Hi,

I am experiencing an issue when trying to run the example code for the Diffusion Policy. I followed the instructions in the [README.md](https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/diffusion_policy/README.md) to:

- Create a new virtual environment
- Replay the trajectory

However, when I run `train.py`, I encounter an error that I cannot resolve. I have attached a screenshot of the error message.

I am unable to locate where this bug is occurring. Could you please help me troubleshoot this issue?

[image]

Thank you!

StoneT2000, 2024-11-28:
What version of maniskill are you using? Pypi or latest git commit?

And did you make any modifications?

And what is the exact command you ran?

Rui-li023, 2024-11-28:
I followed the Installation section in the Diffusion Policy [README.md](https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/diffusion_policy/README.md) to create a new conda environment and installed the environment:

```bash
conda create -n diffusion-policy-ms python=3.9
conda activate diffusion-policy-ms
pip install -e .
```

I replayed the trajectory using the following commands:

```bash
python -m mani_skill.utils.download_demo "PickCube-v1"
python -m mani_skill.trajectory.replay_trajectory \
  --traj-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.h5 \
  --use-first-env-state -c pd_ee_delta_pos -o state \
  --save-traj --num-procs 10
```

Since I just created a new environment, I don't think I've modified anything. As for the code under the `diffusion_policy` folder, it's the same as the `main` branch on GitHub, and I haven't made any changes. The training command is as follows:

```bash
python train.py --env-id PickCube-v1 \
  --demo-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.state.pd_ee_delta_pos.cpu.h5 \
  --control-mode "pd_ee_delta_pos" --sim-backend "cpu" --num-demos 100 --max_episode_steps 200 \
  --total_iters 30000
```

The runtime information is as follows:

```
Loading HDF5 file /home/lirui/.maniskill/demos/StackCube-v1/motionplanning/trajectory.state.pd_ee_delta_pos.cpu.h5
Loaded
Load observations 100 <class 'numpy.ndarray'>
Load actions 100 <class 'numpy.ndarray'>
Total transitions: 10843, Total obs sequences: 10743
number of parameters: 4.43M
number of parameters: 4.43M
Iteration 0, loss: 1.1223094463348389
ERROR: Received the following error from Worker-0: TypeError: 'bool' object is not iterable
ERROR: Shutting down Worker-0.
ERROR: Raising the last exception back to the main process.
Traceback (most recent call last):
  File "/home/lirui/ManiSkill/examples/baselines/diffusion_policy/train.py", line 399, in <module>
    eval_metrics = evaluate(args.num_eval_episodes, ema_agent, envs, device, args.sim_backend)
  File "/home/lirui/ManiSkill/examples/baselines/diffusion_policy/diffusion_policy/evaluate.py", line 35, in evaluate
    obs, rew, terminated, truncated, info = eval_envs.step(action_seq[:, i])
  File "/home/lirui/anaconda3/envs/diffusion-policy-ms/lib/python3.9/site-packages/gymnasium/vector/vector_env.py", line 204, in step
    return self.step_wait()
  File "/home/lirui/anaconda3/envs/diffusion-policy-ms/lib/python3.9/site-packages/gymnasium/vector/async_vector_env.py", line 333, in step_wait
    self._raise_if_errors(successes)
  File "/home/lirui/anaconda3/envs/diffusion-policy-ms/lib/python3.9/site-packages/gymnasium/vector/async_vector_env.py", line 544, in _raise_if_errors
    raise exctype(value)
TypeError: 'bool' object is not iterable
```

What's more, I just did some more experiments. If I downgrade the version using the command `pip install mani_skill==v3.0.0b12`, I can run `train.py` with the above commands normally and get the following output:

```
Loading HDF5 file /home/lirui/.maniskill/demos/PickCube-v1/motionplanning/trajectory.state.pd_ee_delta_pos.cpu.h5
Loaded
Load observations 100 <class 'numpy.ndarray'>
Load actions 100 <class 'numpy.ndarray'>
Total transitions: 7720, Total obs sequences: 7620
number of parameters: 4.39M
number of parameters: 4.39M
Iteration 0, loss: 1.1120742559432983
Evaluated 100 episodes
success_once: 0.0000
return: 0.0000
episode_len: 200.0000
reward: 0.0000
success_at_end: 0.0000
Iteration 1000, loss: 0.023024782538414
Iteration 2000, loss: 0.013510224409401417
Iteration 3000, loss: 0.010976266115903854
Iteration 4000, loss: 0.008350854739546776
```

closed 2024-11-28 by StoneT2000

StoneT2000, 2024-11-28:
@Rui-li023 the latest version 3.0.0b15 has the fixes for this up now, please upgrade to that to get the changes

PR #723 (2024-11-28): [BugFix] Fix bug with record episode wrapper for single CPU envs
Closes #722