mani-skill/ManiSkill #726: [Question] Checking if SRDF file is loading in & working properly

CreativeNick, 2024-11-30:
Building off this issue regarding self-collision joint issues (https://github.com/haosulab/ManiSkill/issues/571), I was wondering if I'm building my SRDF file properly. 

I followed the same general format as the [`panda_pinch.srdf`](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/assets/robots/panda/panda_pinch.srdf) file in `/mani_skill/assets/robots/panda`.

I also made sure the .srdf file is the same exact name as the .urdf file (apart from the filetype of course) and is in the same folder location. I double-checked to also make sure that the `load_articulation` function in [`base_agent.py`](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/agents/base_agent.py) as well as the `load` function in [`urdf_loader.py`](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/building/urdf_loader.py) were both working & unchanged from the original ManiSkill repo. It says in `urdf_loader.py` that "If srdf_file is None, it defaults to the ".srdf" file with the same as the urdf file," so I'm not sure if the issue stems from my `bimanual_allegro.py` file, my .srdf file, or something else entirely.

I've attempted editing and adding debug statements in `urdf_loader.py` and `scene_builder.py`, but they don't seem to be involved in the process of viewing my custom ycb_env environment when I run `python env_utils/view_ycb_env.py` - as in, I could delete all the code in `mani_skill/agents/base_agent.py` and I can run that view ycb env command. I'm not sure if this is the main reason why it's not detecting my SRDF file.

Here is the link to my [SRDF file](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/assets/urdf/ur5e_allegro/robots/dual_ur5e_allegro_inertia_changed.srdf) as well as my [`bimanual_allegro.py`](https://github.com/CreativeNick/SimToReal/blob/main/bimanual-sapien/bimanual_allegro.py) file (which contains the urdf_path).

Thank you!

StoneT2000, 2024-11-30:
Did you do a local pip install? Or did you pip install the pypi package?

CreativeNick, 2024-12-08:
@StoneT2000 Should I uninstall and re-install maniskill via local pip install?

CreativeNick, 2024-12-16:
Uninstalled and re-installed mani_skill and also removed and re-cloned the github repo, the `scene_builder.py` file and other files seem to actually be used now.

closed 2024-12-16 by CreativeNick