mani-skill/ManiSkill #732: [Question] How to manually apply force to an actor

hesic73, 2024-12-03:
I noticed that in  [penspin](https://penspin.github.io/), a random force is applied to an actor at specific simulation steps to serve as a perturbation, which helps improve the robustness of the learned policy. I'm wondering if ManiSkill provides a similar mechanism or an alternative way to manually apply a force to an actor during simulation. Thanks!

https://github.com/HaozhiQi/penspin/blob/f6f2c87ed6a427edd8929751e8feb4bc1744b197/penspin/tasks/allegro_hand_hora.py#L714-L725

```python
def update_rigid_body_force(self):
        if self.force_scale > 0.0:
            self.rb_forces *= torch.pow(self.force_decay, self.dt / self.force_decay_interval)
            # apply new forces
            obj_mass = [self.gym.get_actor_rigid_body_properties(env, self.gym.find_actor_handle(env, 'object'))[0].mass for env in self.envs]
            obj_mass = to_torch(obj_mass, device=self.device)
            prob = self.random_force_prob_scalar
            force_indices = (torch.less(torch.rand(self.num_envs, device=self.device), prob)).nonzero()
            self.rb_forces[force_indices, self.object_rb_handles, :] = torch.randn(
                self.rb_forces[force_indices, self.object_rb_handles, :].shape,
                device=self.device) * obj_mass[force_indices, None] * self.force_scale
            self.gym.apply_rigid_body_force_tensors(self.sim, gymtorch.unwrap_tensor(self.rb_forces), None, gymapi.ENV_SPACE)
```

StoneT2000, 2024-12-03:
It does, I just need to add it. I can do it today/tomorrow.

StoneT2000, 2024-12-16:
Apologies for delay, rigid body external forces are now supported on CPU/GPU.

closed 2024-12-16 by StoneT2000

StoneT2000, 2024-12-16:
https://github.com/haosulab/SAPIEN/releases you need to install the latest wheel in here however to get that feature

hesic73, 2024-12-17:
I've recently been thinking about alternatives to RL, which also involve applying forces. Thank you so much!

PR #752 (2024-12-16): [Feature] Support applying external forces on objects
closes #732