mani-skill/ManiSkill #733: Request for Model Checkpoint Download

yeknafar, 2024-12-03:
First of all, I would like to thank you for your amazing work 

I want to use the pre-trained wieghts. Could you please provide me with the location or a link where I can download the checkpoint of the model?

Thank you in advance for your help.

StoneT2000, 2024-12-03:
Which pretrained checkpoints/weights?

yeknafar, 2024-12-07:
any model that has been trained so far and the results are in w&b.

StoneT2000, 2024-12-08:
For the PPO baselines which we also use to generate demonstrations you can find them here: https://huggingface.co/datasets/haosulab/ManiSkill_Demonstrations

We upload pre-generated demonstrations there + the weights of the policy used to generate them. 

You can also download demos+weights via the CLI `python -m mani_skill.utils.download_demo all`

Apologies for the lack of a documentation page for pretrained weights, we will add links to that in our docs later, for now u can also see the demonstrations docs for how to download task specfic demos (some of which come with PPO pretrained weights).
https://maniskill.readthedocs.io/en/latest/user_guide/datasets/demos.html#demonstrations

closed 2024-12-16 by StoneT2000