mani-skill/ManiSkill #734: stable-baselines3 immitation lab could not collection data with rollout function

AcademeMaster, 2024-12-03:
"Hi, I am currently utilizing the stable-baselines3 imitation lab, but I am encountering issues when attempting to integrate it with the ManiSkillVectorEnv wrapper and the ManiSkillSb3VectorEnv. Has anyone else tried this combination before? If so, I would greatly appreciate any help or advice you could offer. Thank you in advance."

##code

```
import numpy as np
from imitation.algorithms.adversarial.gail import GAIL
from imitation.data import rollout
from imitation.data.wrappers import RolloutInfoWrapper
from imitation.rewards.reward_nets import BasicRewardNet
from imitation.util.networks import RunningNorm
from mani_skill.vector.wrappers.gymnasium import ManiSkillVectorEnv
from mani_skill.vector.wrappers.sb3 import ManiSkillSB3VectorEnv

from stable_baselines3.common.evaluation import evaluate_policy
from stable_baselines3.ppo import MlpPolicy

from SnnGail.SpikingNet import SnnDiscriminatorNet
from stable_baselines3.common.env_checker import check_env

import gymnasium as gym
from stable_baselines3 import PPO
from stable_baselines3.common.utils import set_random_seed
from stable_baselines3.common.vec_env import SubprocVecEnv
import os
import math
import gymnasium as gym

from stable_baselines3 import PPO
from stable_baselines3.common.env_util import make_vec_env
from sympy import det
from mani_skill.utils import gym_utils
from mani_skill.utils.wrappers.record import RecordEpisode

# from mani_skill.vector.wrappers.sb3 import ManiSkillSB3VectorEnv

import math

from SnnGail.Sb3Net import CustomActorCriticPolicy

# ppo_pickcube18 64 128 64
# 定义模型路径
expert_model_path = "./model/ppo_pickcube_ann1"

# 自动学习率
def create_linear_lr_schedule(initial_value, min_lr=1e-8):
    def lr_lambda(progress_remaining):
        return max(progress_remaining * initial_value, min_lr)
    return lr_lambda

ms3_vec_env = gym.make(
    "PickCube-v1",
    num_envs=1,
    robot_uids="ur5",
    obs_mode="state",  # there is also "state_dict", "rgbd", ...
    reward_mode="normalized_dense",  # there is also "sparse", "dense", ...
    control_mode="pd_ee_delta_pose",  # there is also "pd_joint_delta_pos", ...
    render_mode="rgb_array",
    max_episode_steps=100
)

vec_env = ManiSkillVectorEnv(ms3_vec_env)

# # 检查模型是否存在
# if os.path.exists(f"{expert_model_path}.zip"):
#     print(f"加载模型 {expert_model_path}")
#     expert_model = PPO.load(expert_model_path, env=vec_env)  # 加载模型
# else:
#     print("未找到模型，创建新模型")
# print("开始训练")
expert_model = PPO(
    policy=MlpPolicy,
    env=vec_env,
    learning_rate=1e-4,
    n_steps=2048,
    batch_size=64,
    n_epochs=10,
    gamma=0.99,
    gae_lambda=0.95,
    clip_range=0.2,
    ent_coef=0.0,
    verbose=1,
    seed=42,
)
expert_model.learn(total_timesteps=1)

vec_env= RolloutInfoWrapper(vec_env)

SEED = 100
rollouts_data = rollout.rollout(
    expert_model,
    vec_env,
    rollout.make_sample_until(min_timesteps=None, min_episodes=1),
    rng=np.random.default_rng(SEED),
)

```

StoneT2000, 2024-12-10:
what's the error output?

AcademeMaster, 2024-12-13:
I have sovled this problem ,because we can not use rolloutWrapper.But when I run GAIL example for pickcube task,can not success.

the key code was add in ManiSkillSB3VectorEnv  follow ：
  new_infos[i]["rollout"] = {
      "obs": np.stack(self._obs[i]),
      "rews": np.stack(self._rews[i]),
  }

AcademeMaster, 2024-12-13:
> what's the error output?
I have sovled this problem ,because we can not use rolloutWrapper.But when I run GAIL example for pickcube task,can not success.

the key code was add in ManiSkillSB3VectorEnv follow ：
```
new_infos[i]["rollout"] = {
"obs": np.stack(self._obs[i]),
"rews": np.stack(self._rews[i]),
}

```

StoneT2000, 2024-12-19:
issue resolved in private message

closed 2024-12-19 by StoneT2000