mani-skill/ManiSkill #736: About fisheye camera

XinnWang, 2024-12-05:
Hi, i want to add a fisheye camera, is there any support for this ?

StoneT2000, 2024-12-08:
I think @fbxiang had something way to do it easily, any idea on the best way to do so with SAPIEN?

fbxiang, 2024-12-08:
I have experimented with this. Assuming you want real time rendering, since fisheye cannot be expressed as a perspective transformation, we cannot use rasterization to do it directly. The simplest way is to actually render a normal camera with large fov and high resolution, and then distort it as a postprocessing. Programming it in SAPIEN directly requires quite a bit of deep knowledge about SAPIEN and shader programming, so I actually recommend doing it in torch (or numpy if you are already processing images on the CPU).

XinnWang, 2024-12-09:
@fbxiang thanks so much. we will try this method.

closed 2024-12-10 by StoneT2000

lieyanzhan, 2026-02-03:
> I have experimented with this. Assuming you want real time rendering, since fisheye cannot be expressed as a perspective transformation, we cannot use rasterization to do it directly. The simplest way is to actually render a plain old camera image with large fov and high resolution, and then distort it as a postprocessing. Programming it in SAPIEN directly requires quite a bit of deep knowledge about SAPIEN and shader programming, so I actually recommend doing it in torch (or numpy if you are already processing images on the CPU).

l have the same question with you. l also want to get a fisheye camera in the maniskill3, could you give some ecample about it. thank you very much!