mani-skill/ManiSkill #739: Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath

xjtu19, 2024-12-07:
When I run the command `pytest -vs` in the terminal, there is a notification (not an error):
`
tests/test_envs.py::test_all_envs[PickClutterYCB-v1] Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath
PASSED
tests/test_envs.py::test_all_envs[PickCube-v1] Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath
PASSED
tests/test_envs.py::test_all_envs[PickSingleYCB-v1] Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath
PASSED
`
What dose it mean? And how to settle it?

StoneT2000, 2024-12-08:
@fbxiang ?

fbxiang, 2024-12-08:
It may have something to do with how openvr behaves on Windows (e.g. when Steam is installed, simply checking support for openvr may cause it to talk to Steam). I do not use Windows so I am not sure about the root cause. If it is just a harmless message, I would say just ignore it.

closed 2024-12-10 by StoneT2000