mani-skill/ManiSkill #740: [svulkan2] [error] and IndexError

xjtu19, 2024-12-07:
PS D:\E20220227\C000_ManiSkill> python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human" --shader="rt-fast"
C:\Users\26735\anaconda3\envs\py310\lib\site-packages\sapien\wrapper\pinocchio_model.py:283: UserWarning: pinnochio package is not installed, robotics functionalities will not be available
  warnings.warn(
Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath
Unable to read VR Path Registry from C:\Users\26735\AppData\Local\openvr\openvrpaths.vrpath
[2024-12-07 22:46:27.272] [svulkan2] [error] The selected GPU does not support ray tracing
[2024-12-07 22:46:28.279] [svulkan2] [error] The selected GPU does not support ray tracing
Observation space Dict()
Action space Box(-1.0, 1.0, (8,), float32)
Control mode pd_joint_delta_pos
Reward mode normalized_dense
[2024-12-07 22:46:28.671] [svulkan2] [error] The selected GPU does not support ray tracing
[2024-12-07 22:46:28.734] [svulkan2] [error] The selected GPU does not support ray tracing
[2024-12-07 22:46:28.803] [svulkan2] [error] The selected GPU does not support ray tracing
Traceback (most recent call last):
  File "C:\Users\26735\anaconda3\envs\py310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\26735\anaconda3\envs\py310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\E20220227\C000_ManiSkill\mani_skill\examples\demo_random_action.py", line 128, in <module>
    main(parsed_args)
  File "D:\E20220227\C000_ManiSkill\mani_skill\examples\demo_random_action.py", line 103, in main
    viewer = env.render()
  File "C:\Users\26735\anaconda3\envs\py310\lib\site-packages\gymnasium\core.py", line 471, in render
    return self.env.render()
  File "C:\Users\26735\anaconda3\envs\py310\lib\site-packages\gymnasium\wrappers\order_enforcing.py", line 70, in render
    return self.env.render(*args, **kwargs)
  File "D:\E20220227\C000_ManiSkill\mani_skill\envs\sapien_env.py", line 1309, in render
    return self.render_human()
  File "D:\E20220227\C000_ManiSkill\mani_skill\envs\sapien_env.py", line 1244, in render_human
    self._viewer.render()
  File "C:\Users\26735\anaconda3\envs\py310\lib\site-packages\sapien\utils\viewer\viewer.py", line 225, in render
    self.window.render(self.render_target, ui_windows)
  File "C:\Users\26735\anaconda3\envs\py310\lib\site-packages\sapien\utils\viewer\render_window.py", line 110, in exposure
    return self.window.get_camera_property_float("exposure")
IndexError: invalid unordered_map<K, T> key

xjtu19, 2024-12-07:
My platforn is windows 10

StoneT2000, 2024-12-08:
Can you try

```
python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human"
```
first?

xiaojing0418, 2024-12-08:
> Can you try
> 
> ```
> python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human"
> ```
> 
> first?

It's OK now! Thanks a lot. 
PS. In the pop-up SAPIEN viewer, when I select 'rt' in the **Shader** drop-down box of the **Render** toolbar, the SAPIEN collapsed immediately. It's OK though.

StoneT2000, 2024-12-08:
Most likely some issue with ray tracing on windows, not sure however, @fbxiang might know

fbxiang, 2024-12-09:
Yes, I think there are many issues with ray tracing on Windows. The Windows version is extremely experimental at this point. Please consider switching to Linux if anything you want does not work.

closed 2024-12-10 by StoneT2000