mani-skill/ManiSkill #743: Questions about Warp version

Lyy-iiis, 2024-12-13:
Hello, when running maniskill2, I find a interesting soft-body env called Pinch-v0. However, when I try the env, `ImportError: cannot import name 'DenseVolume' from 'warp.sim.model' (/root/miniconda3/envs/maniskill/lib/python3.9/site-packages/warp/sim/model.py)` happens. 
I already install the correct version in https://github.com/haosulab/ManiSkill2-docs/blob/main/warp_maniskill/VERSION.md . But the error still happens. Can you tell me the way to resolve this, thank you!

StoneT2000, 2024-12-15:
@fbxiang @Rabbit-Hu

fbxiang, 2024-12-15:
Could you confirm this only happens to the Pinch env or does it happen for all envs? If it happens for all envs, I think you may have installed a standalone warp yourself instead of using the one bundled with ManiSkill2, where we implemented the dense volume.

Lyy-iiis, 2024-12-17:
I fix this by installing correct version of maniskill (it seems both `mani-skill2` and `mani_skill2` exists), now it can work properly.

closed 2024-12-17 by Lyy-iiis