mani-skill/ManiSkill #744: ManiSkill3-Learn

AcademeMaster, 2024-12-13:
"Hello, may I ask if there are any plans from the official team to update the ManiSkill-Learn library to be compatible with ManiSkill3? If such plans are not available at the moment, how should I modify and adapt the library to work in the ManiSkill3 environment?"

StoneT2000, 2024-12-15:
I generally would recommend just using the baselines we have now in the examples/baselines folder, which provide PPO, SAC, TDMPC2, and several common imitation learning algorithms baselines. They are written to be independent of each other for easier readability so it is in a different style to ManiSkill-Learn which is more modularized.

Otherwise maybe @xuanlinli17 may have an answer, he maintained that library previously.

xuanlinli17, 2024-12-16:
Unfortunately I don't have time to write ManiSkill3-Learn. You can use the existing algorithms from the ManiSkill repo or adopt from ManiSkill2-Learn.

AcademeMaster, 2024-12-16:
@StoneT2000 Thanks,I will try.

closed 2024-12-16 by StoneT2000