mani-skill/ManiSkill #746: GPU Devices

nicholscrawford, 2024-12-13:
Hi, 

I'm using maniskill and am unclear on the intended use in terms of which devices I'm supposed to have data on. It's returning 
``` obs, reward, terminated, truncated, info = self.env.step(action) ```
But terminated and truncated are on different GPUs. Is this intended? What's the intended workflow? I'd expect something like computing my network stuff on one GPU, and having the environments return on the same one. 

It feels silly to write:
```
obs, info = self.env.reset(options={'env_idx': torch.nonzero(torch.logical_or(terminated, truncated.to(terminated.device))).squeeze()})
```

StoneT2000, 2024-12-15:
This doesn't seem right, what does `nvidia-smi` say?

Generally in GPU sims everything is on one GPU per python process, we shouldn't have any code that allows for tensors to be on different devices.

nicholscrawford, 2024-12-16:
Well I'm working on reproducing it in a simpler script, but relatedly this is producing an error:
```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.120                Driver Version: 550.120        CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090        Off |   00000000:01:00.0 Off |                  Off |
|  0%   43C    P8             29W /  480W |    3379MiB /  24564MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090        Off |   00000000:02:00.0  On |                  Off |
|  0%   38C    P3             52W /  480W |    8280MiB /  24564MiB |     10%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```
Here's a python snippet with a similar problem:
```python
import gymnasium as gym
import mani_skill.envs

import os
os.environ['CUDA_VISIBLE_DEVICES'] = '0'

env=gym.make("PickSingleYCB-v1", num_envs=10)
obs = env.reset()

res = env.step(action)
print(env.unwrapped.device)
for item in res:
    if type(item) == torch.Tensor: print(item.device)
    
import mani_skill
print(mani_skill.__version__)
print(gym.__version__)
```

outputs:

```
cuda
cuda:1
cuda:1
cuda:1
cuda:1
3.0.0b14
0.29.1
```

nicholscrawford, 2024-12-16:
Okay looks like this is a good reproduction:

```python
import gymnasium as gym
import mani_skill.envs

import os
os.environ['CUDA_VISIBLE_DEVICES'] = '0'

import torch
print(torch.cuda.device_count())  
print(torch.cuda.get_device_name(0))  

import torch
torch.cuda.set_device(0)  # Explicitly set GPU 0

env=gym.make("PickSingleYCB-v1", num_envs=10)
obs = env.reset()

res = env.step(action)
print(env.unwrapped.device)
for item in res:
    if type(item) == torch.Tensor: print(item.device)
    
import mani_skill
print(mani_skill.__version__)
print(gym.__version__)
```

outputs:
```
1
NVIDIA GeForce RTX 4090
cuda
cuda:1
cuda:1
cuda:1
cuda:0
3.0.0b14
0.29.1
```

StoneT2000, 2024-12-16:
Not immediately sure what the issue is. Could you try upgrading maniskill to the latest version and try again?

And I think one possible workaround is to instead run your script as so

```bash
export CUDA_VISIBLE_DEVICES=1 python script.py
```

nicholscrawford, 2024-12-16:
It is the latest version, one of the version markers just isn't updated. 

[https://github.com/haosulab/ManiSkill/blob/main/mani_skill/__init__.py#L6C1-L6C25](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/__init__.py#L6C1-L6C25)

Although I tried moving the os environment variable export to before the mani_skill import, and that seems to have fixed things. Broadly speaking, is multi-gpu supported at all?

StoneT2000, 2024-12-16:
You can run separate maniskill instances using different gpus in different processes. For some distributed training workloads one would need to investigate a bit more into pytorch distributed I think. It is the solution isaac lab uses for multi-gpu training although I am not sure how straight forward it is at the moment.

closed 2024-12-17 by nicholscrawford