mani-skill/ManiSkill #747: 3D diffusion policy baseline

ggggfff1, 2024-12-14:
I would like to inquire whether there are plans to develop another diffusion policy method as a baseline, such as the  [3D Diffusion Policy (DP3)3D Diffusion Policy (DP3)](https://3d-diffusion-policy.github.io/)

StoneT2000, 2024-12-15:
We don't have plans at the moment to add DP3 as a tested baseline. However we are open to pull requests to help merge a baseline like that in! DP3 is certainly a nice baseline we would love to have. I can provide guidance on the necessary requirements for new baselines in this repo (e.g. what metrics to report, wandb logs etc.).

ggggfff1, 2024-12-24:
I have encountered a problem while trying to use point cloud data as observations to complete the PickCube-v1 task. The point cloud does not capture the target position, which hinders the successful completion of the task.

My question is: How can we ensure that the target position is observed when using point cloud data as observation input (e.g., in the PickCube task)?

StoneT2000, 2024-12-26:
What do you mean it is not captured? Do you have an example? Is target position referring to the green goal? If so the target position is part of the observation dict (see obs["extra"])

ggggfff1, 2024-12-30:
[image]

For example, in the PickCube-v1 task, the original point cloud does not include the target position. For pure point cloud processing algorithms (such as DP3), the amount of point cloud information here is not sufficient to complete the task. However, manually completing the point cloud from the information in obs ['extra '] may be a solution.

But the best way is still to obtain a point cloud of the virtual object (target location, green point in the following figure)
[image]

ruoxianglee, 2025-01-14:
> I would like to inquire whether there are plans to develop another diffusion policy method as a baseline, such as the [3D Diffusion Policy (DP3)3D Diffusion Policy (DP3)](https://3d-diffusion-policy.github.io/)

Hi, did you finish implementing DP3 in ManiSkill? I just post an [issue](https://github.com/haosulab/ManiSkill/issues/799) about this topic. I trained DP3 for PickCube task in ManiSkill, but got very low success rate. Do you have any insights for this issue? Thanks. Hope to discuss further with you on this topic. My email is ruoxiang.li@my.cityu.edu.hk.