mani-skill/ManiSkill #755: Add task language instructions

StoneT2000, 2024-12-17:


rechim25, 2025-03-19:
Hi, I'm curios if there are any updates on this enhancement.

Currently working with VLAs so would be very convenient to have this features :)

StoneT2000, 2025-03-19:
This is extremely easy and hard to do. The easy part is coming with language descriptors for each task.

The hard part is considering how we want to support this.

For example a simple way is to add a function called `def get_language_description()` and make that an official part of the `BaseEnv` class. However since we work with GPU simulations sometimes maybe you have a different language description in each parallel environment

For example in PickSingleYCB the object to pick up could be different in each parallel sim. Then this language function might have to output a batch of strings? Or a batch of already tokenized strings?

I am not sure what is the best way to do so.

rechim25, 2025-03-20:
Indeed @StoneT2000 .

I think it would make sense to put the **language description as part of the observation dict** (in 'extra') and let each particular env implementation decide how to construct it (in '_get_obs_extra()'). This way it's possible to:
- Add spatial cues (e.g., "up", "left", etc.) to the language description based on privileged info such as object position etc. 
- Change the language description throughout the episode (if needed). For example, more complex envs might want to reconstruct the language description based on certain triggers.
- We don't force all envs to implement the language description.

The output format could be a **torch.Tensor with utf-8 byte-encoded descriptions** for each parallel env. We define a MAX_LENGTH length for all language description and pad the remaining. This way we can batch them.

To decode the strings from the obs we can add an utility function for the user.

Curios to know what you think about this.

StoneT2000, 2025-03-20:
This might conflict since all the data in extra is generally fed directly to policy networks (like the actor/critic in PPO) which do not use language by default. Users will have to manually remove "language" from the observations and we provide observation flattening as a neat feature which most people use and thus even harder to remove.

Another potential option is to add a new modality of observation. We have something called the `env.unwrapped.obs_struct` (poorly named probably) that is a dataclass config detailing what the user wants to generated in their observations.

https://github.com/haosulab/ManiSkill/blob/3958362e2abb1322cb51b929cdc89f42663e4a2b/mani_skill/envs/utils/observations/__init__.py#L6-L31

Basically a bunch of bools dictating what observation data should be returned from env.step and env.reset calls.

Perhaps we can add a new section called language so 

```
@dataclass
class LanguageObsStruct:
    ??: bool

@dataclass
class ObservationModeStruct:
    """A dataclass describing what observation data is being requested by the user"""

    state_dict: bool
    """whether to include state data which generally means including privileged information such as object poses"""
    state: bool
    """whether to include flattened state data which generally means including privileged information such as object poses"""
    visual: CameraObsTextures
    """textures to capture from cameras"""
    language: LanguageObsStruct

```

then for custom tasks users can either do whatever they want (e.g. override the observation function), or use our system accordingly and define a new def _get_obs_language(...) function that we then try to help them pad/batch?

StoneT2000, 2025-03-20:
My only other concern is adding such a feature might just be futile. I find a lot of other benchmarks tend to just have some of their own defined function that returns a string and their inference code parses it in different ways

yqi19, 2025-06-07:
Hi Author, I didn't quite get the previous part. Why is it be the futile? For example, in the PickClutterEnv, the language can be as a goal interpretation, and different language indicate different success conditions. I wonder how would you consider this？

StoneT2000, 2025-06-07:
Futile in the sense that such an API is very likely to change quickly and I think it is better for users to come up with their own way of adding language. LIBERO does it one way, others do it other ways, and none have considered batched language descriptors that I proposed earlier. Given the uncertainty and many possible choices I recommend making your own setup. Especially since MS3 does not benchmark VLAs of any kind at the moment, it's hard to do many tests.

If you really want to use language now you can just use 

environment_class.__doc__

most environments have doc strings explaining what the task is and what the success conditions and randomizations are.

yqi19, 2025-06-07:
Thanks for the detailed explanations. However, adding environment.__doc__ is adding permanent language instructions rather than language variants. The same task can have different success condition based on randomly sampled language instructions, e.g. place wine on the right/middle/left rack location. So in my understanding, they should be initialized when each episode starts. So it seems that it still needs to be saved as part of the observations. Or are there any other better solutions? Thanks!

yqi19, 2025-06-07:
Besides, may be out lang in the info can also be an option?

StoneT2000, 2025-06-11:
I think adding language to the info object is possible, but it is not good practice because the info object is for end user defined data, non standardized.

ManiSkill framework intends to keep as much standardization as possible. My suggestion from before about adding a language observation mode still is a good option. Language observations should be in the obs object. But the complication occurs if you consider batching

One option is to have several language modalities. A string version and a tokenized version maybe. Tokenized one being more efficient when running parallel sim.