mani-skill/ManiSkill #757: why can't gripper  pick the cube

JuilieZ, 2024-12-18:
I replay this trajectory but the grip cannot pick up the cube,I have changed the stiffness,damping,effort and friction,but make no sense

https://github.com/user-attachments/assets/010e911e-b904-4106-becd-ae25df1fd6e9

StoneT2000, 2024-12-19:
Is this your custom robot or one of the prebuilt ones?

JuilieZ, 2024-12-19:
> 这是您定制的机器人还是预制的机器人之一？

custom robot,more specifically,it's realman arm and inspire gripper(EG2)

JuilieZ, 2024-12-19:
Another question I can't understand : I set the friction (of the gripper of panda) to 0 , the gripper  pick up the cube still , which is abnormal
`        _materials=dict(
            gripper=dict(static_friction=0, dynamic_friction=0, restitution=0.0)
        ),`

StoneT2000, 2024-12-23:
can you share the URDF and the agent class you created? Could be something not defined well.

JuilieZ, 2024-12-23:
> can you share the URDF and the agent class you created? Could be something not defined well.

of course,here is my robot urdf
`<?xml version="1.0" encoding="utf-8"?>

<robot
  name="rm_75_6f_description">
  
  <link name="base_link">
    <inertial>
      <origin  xyz="0.00043296 3.5291E-05 0.059942" rpy="0 0 0" />
      <mass value="0.84105" />
      <inertia ixx="0.0017261" ixy="2.5275E-06" ixz="3.6769E-05" iyy="0.0017099" iyz="-1.68E-06" izz="0.00090402" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="/home/server/ManiSkill/mani_skill/assets/robots/rm75/meshes/base_link.STL" />
      </geometry>
      <material name="">
        <color rgba="1 1 1 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="/home/server/anaconda3/envs/ManiSkill/lib/python3.12/site-packages/mani_skill/assets/robots/rm75/meshes/base_link.STL" />
      </geometry>
    </collision>
  </link>
  <link name="Link1">
    <inertial>
      <origin xyz="-2.1376E-07 0.025186 -0.021108" rpy="0 0 0" />
      <mass value="0.59356" />
      <inertia ixx="0.0012661" ixy="-9.8019E-09" ixz="-1.2951E-08" iyy="0.00056135" iyz="-0.00021122" izz="0.0011817" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rm75/meshes/Link1.STL" />
      </geometry>
      <material name="">
        <color rgba="1 1 1 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rm75/meshes/Link1.STL" />
      </geometry>
    </collision>
  </link>
  <joint name="joint1" type="revolute">
    <safety_controller k_position="100.0" k_velocity="40.0" soft_lower_limit="-3.107" soft_upper_limit="3.107"/>
    <origin xyz="0 0 0.2405" rpy="-1.5708 0 0" />
    <parent link="base_link" />
    <child  link="Link1" />
    <axis   xyz="0 -1 0" />
    <limit lower="-3.107" upper="3.107" effort="80" velocity="1" />
    <dynamics D="1" K="7000" damping="0.003" friction="0.0" mu_coulomb="0" mu_viscous="16"/>
  </joint>
  <link name="Link2">
    <inertial>
      <origin xyz="4.1111E-07 0.011078 0.076129" rpy="0 0 0" />
      <mass value="0.43285" />
      <inertia ixx="0.0012584" ixy="5.6315E-09" ixz="1.3348E-09" iyy="0.0012225" iyz="0.000279" izz="0.00031747" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
[... 313 lines cut ...]
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
      <material  name="">
        <color   rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin  xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh  filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint4"  type="revolute">
    <origin   xyz="0 -0.022341 0.035565"  rpy="0 0 0" />
    <parent   link="eg2_Link3" />
    <child    link="eg2_Link4" />
    <axis     xyz="1 0 0" />
    <limit    lower="-0.01"   upper="0.8" effort="1e10"     velocity="0.2" />
    <mimic joint="eg2_joint2"  multiplier="1"  offset="0" />
    <dynamics damping="0.3"/>
  </joint>
</robot>`
(Actually I've solved this issue by setting the parameters(stiffness,damping,friction) to a very very large value,which I think is not reasonable.Really hope you can propose other better solution,thank u very much

JuilieZ, 2024-12-23:
> can you share the URDF and the agent class you created? Could be something not defined well.
here is my robot urdf:
<?xml version="1.0" encoding="utf-8"?>

<robot
  name="rm_75_6f_description">
  
  <link name="base_link">
    <inertial>
      <origin  xyz="0.00043296 3.5291E-05 0.059942" rpy="0 0 0" />
      <mass value="0.84105" />
      <inertia ixx="0.0017261" ixy="2.5275E-06" ixz="3.6769E-05" iyy="0.0017099" iyz="-1.68E-06" izz="0.00090402" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="/home/server/ManiSkill/mani_skill/assets/robots/rm75/meshes/base_link.STL" />
      </geometry>
      <material name="">
        <color rgba="1 1 1 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="/home/server/anaconda3/envs/ManiSkill/lib/python3.12/site-packages/mani_skill/assets/robots/rm75/meshes/base_link.STL" />
      </geometry>
    </collision>
  </link>
  <link name="Link1">
    <inertial>
      <origin xyz="-2.1376E-07 0.025186 -0.021108" rpy="0 0 0" />
      <mass value="0.59356" />
      <inertia ixx="0.0012661" ixy="-9.8019E-09" ixz="-1.2951E-08" iyy="0.00056135" iyz="-0.00021122" izz="0.0011817" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rm75/meshes/Link1.STL" />
      </geometry>
      <material name="">
        <color rgba="1 1 1 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rm75/meshes/Link1.STL" />
      </geometry>
    </collision>
  </link>
  <joint name="joint1" type="revolute">
    <safety_controller k_position="100.0" k_velocity="40.0" soft_lower_limit="-3.107" soft_upper_limit="3.107"/>
    <origin xyz="0 0 0.2405" rpy="-1.5708 0 0" />
    <parent link="base_link" />
    <child  link="Link1" />
    <axis   xyz="0 -1 0" />
    <limit lower="-3.107" upper="3.107" effort="80" velocity="1" />
    <dynamics D="1" K="7000" damping="0.003" friction="0.0" mu_coulomb="0" mu_viscous="16"/>
  </joint>
  <link name="Link2">
    <inertial>
      <origin xyz="4.1111E-07 0.011078 0.076129" rpy="0 0 0" />
      <mass value="0.43285" />
      <inertia ixx="0.0012584" ixy="5.6315E-09" ixz="1.3348E-09" iyy="0.0012225" iyz="0.000279" izz="0.00031747" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://rm75/meshes/Link2.STL" />
[... 311 lines cut ...]
      <origin   xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
      <material  name="">
        <color   rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin  xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh  filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint4"  type="revolute">
    <origin   xyz="0 -0.022341 0.035565"  rpy="0 0 0" />
    <parent   link="eg2_Link3" />
    <child    link="eg2_Link4" />
    <axis     xyz="1 0 0" />
    <limit    lower="-0.01"   upper="0.8" effort="1e10"     velocity="0.2" />
    <mimic joint="eg2_joint2"  multiplier="1"  offset="0" />
    <dynamics damping="0.3"/>
  </joint>
</robot>

JuilieZ, 2024-12-23:
> can you share the URDF and the agent class you created? Could be something not defined well.

  
  <link name="eg2_hand_tcp"/>
  <joint name="eg2_hand_tcp_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0.15"/>
    <parent link="eg2_base_link"/>
    <child link="eg2_hand_tcp"/>
  </joint>

  <link name="eg2_base_link">
    <inertial>
      <origin   xyz="-2.6792E-06 1.4343E-05 0.040955"    rpy="0 0 0" />
      <mass     value="0.075349" />
      <inertia  ixx="6.9695E-05"    ixy="-2.8955E-12"   ixz="1.2653E-08"   iyy="6.0426E-05"   iyz="2.2585E-08"   izz="1.9195E-05" />
    </inertial>
    <visual>
      <origin   xyz="0 0 0"    rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_base_link.STL" />
      </geometry>
      <material name="">
        <color  rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_base_link.STL" />
      </geometry>
    </collision>
  </link>
  <joint name="rm_hand_joint" type="fixed">
    <parent link="Link7"/>
    <child link="eg2_base_link"/>
    <origin rpy="0 0 0" xyz="0 0 0.015"/>
  </joint>
  <link name="eg2_Link1">
    <inertial>
      <origin  xyz="0.0001 0.0085832 0.013664"  rpy="0 0 0" />
      <mass    value="0.0024857" />
      <inertia ixx="5.8907E-07"  ixy="1.3343E-19" ixz="4.2984E-19"  iyy="5.2237E-07"  iyz="-2.5787E-07"   izz="2.7386E-07" />
    </inertial>
    <visual>
      <origin  xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh  filename="package://rm75/meshes/eg2_Link1.STL" />
      </geometry>
      <material  name="">
        <color   rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin    xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh    filename="package://rm75/meshes/eg2_Link1.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint1"   type="revolute">
    <origin   xyz="0 -0.04 0.09"   rpy="0 0 0" />
    <parent   link="eg2_base_link" />
    <child    link="eg2_Link1" />
    <axis     xyz="1 0 0" />
    <limit    lower="0"   upper="0.8"   effort="1e10"    velocity="0.2" />
    <mimic joint="eg2_joint2"  multiplier="1"  offset="0" />    
    <dynamics damping="0.3"/>
  </joint>
  <link   name="eg2_Link2">
    <inertial>
      <origin   xyz="9.9997E-05 0.011846 0.011367"    rpy="0 0 0" />
      <mass     value="0.0044927" />
      <inertia  ixx="4.099E-07"   ixy="-7.8736E-14"  ixz="-7.8248E-14"  iyy="4.2946E-07"  iyz="-3.8654E-08"  izz="1.6162E-07" />
    </inertial>
    <visual>
      <origin   xyz="0 0 0"    rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_Link2.STL" />
      </geometry>
      <material  name="">
        <color   rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin    xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh    filename="package://rm75/meshes/eg2_Link2.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint2" type="revolute">
    <origin   xyz="0 0.022341 0.035565"   rpy="0 0 0" />
    <parent   link="eg2_Link1" />
    <child    link="eg2_Link2" />
    <axis     xyz="-1 0 0" />
    <limit    lower="-0.01"    upper="0.8"   effort="1e10"     velocity="0.2"/>
    
    <dynamics damping="0.3"/>
  </joint>
  <link  name="eg2_Link3">
    <inertial>
      <origin  xyz="0.0001 -0.0085832 0.013664"   rpy="0 0 0" />
      <mass    value="0.0024857" />
      <inertia   ixx="5.8907E-07"  ixy="-3.3324E-19"  ixz="3.1193E-19" iyy="5.2237E-07" iyz="2.5787E-07"  izz="2.7386E-07" />
    </inertial>
    <visual>
      <origin   xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_Link3.STL" />
      </geometry>
      <material name="">
        <color  rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin   xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh  filename="package://rm75/meshes/eg2_Link3.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint3" type="revolute">
    <origin  xyz="0 0.04 0.09"  rpy="0 0 0" />
    <parent  link="eg2_base_link" />
    <child   link="eg2_Link3" />
    <axis    xyz="-1 0 0" />
    <limit   lower="-0.01"   upper="0.8"    effort="1e10"     velocity="0.2" />
    <mimic joint="eg2_joint2"  multiplier="1"  offset="0" />
    <dynamics damping="0.3"/>
  </joint>
  <link   name="eg2_Link4">
    <inertial>
      <origin  xyz="9.9999E-05 -0.011846 0.011367"  rpy="0 0 0" />
      <mass    value="0.0044926" />
      <inertia  ixx="4.099E-07"  ixy="-1.4237E-13"   ixz="-2.483E-13"  iyy="4.2946E-07"  iyz="3.8655E-08"  izz="1.6162E-07" />
    </inertial>
    <visual>
      <origin   xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh   filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
      <material  name="">
        <color   rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
    <collision>
      <origin  xyz="0 0 0"  rpy="0 0 0" />
      <geometry>
        <mesh  filename="package://rm75/meshes/eg2_Link4.STL" />
      </geometry>
    </collision>
  </link>
  <joint  name="eg2_joint4"  type="revolute">
    <origin   xyz="0 -0.022341 0.035565"  rpy="0 0 0" />
    <parent   link="eg2_Link3" />
    <child    link="eg2_Link4" />
    <axis     xyz="1 0 0" />
    <limit    lower="-0.01"   upper="0.8" effort="1e10"     velocity="0.2" />
    <mimic joint="eg2_joint2"  multiplier="1"  offset="0" />
    <dynamics damping="0.3"/>
  </joint>

JuilieZ, 2024-12-23:
> can you share the URDF and the agent class you created? Could be something not defined well.
[rm75.zip](https://github.com/user-attachments/files/18224821/rm75.zip)
it seem failing to show the urdf,I compress it to .zip

StoneT2000, 2024-12-26:
Do you have the reset of the assets for the URDF? Moreover do you have an example base agent class implementation

JuilieZ, 2024-12-31:
> Do you have the reset of the assets for the URDF? Moreover do you have an example base agent class implementation

Thx for ur reply.I've figured out the problem.Actually ,it's all because of the structure of my gripper.The opposite direction of rotation in the 4 'revolute' joints results in the failure when picking up.

StoneT2000, 2025-01-01:
Sounds great! Closing for now

closed 2025-01-01 by StoneT2000