mani-skill/ManiSkill #758: set_qpos doesn't work when running in GPU parallelization

TritiumR, 2024-12-19:
Thanks for your amazing work and implementation for easy GPU parallelization.

I recently switched to GPU parallelization and encounter the bug that I can not reset the robots to original qpos by 
```
"self.agent = self.env.unwrapped.agent
self.agent.robot.set_qpos(current_joint_angles)"
```

It works well for the CPU simulation, as shown in the first video. But there is no reset behavior shown in the GPU simulation in the second video.

And the weird thing is that when I check the qpos by "self.agent.robot.get_qpos()" after setting it, the returned values show that the robots **are** indeed reset to the same original qpos. 

Is there any bug with the set_qpos() in the GPU simulation. Are there any quick workaround that can help? (e.g., restore the whole environment to the previously saved one).

Thanks so much!

https://github.com/user-attachments/assets/de8f614b-0b74-41f8-acba-9229fe09e927

https://github.com/user-attachments/assets/118d7099-5798-4831-8bef-37775608e4d5

StoneT2000, 2024-12-19:
where are you setting this value? With GPU simulation anything written outside of the initialize episode function to set gpu data should come with a gpu apply all call

TritiumR, 2024-12-19:
I am setting outside of the initialize function, before env.step(action). Should I manually call the gpu apply all function or the set_qpos could handle it internally?

TritiumR, 2024-12-19:
A quick question: should I just pass qpos with batch size one to the set_qpos() instead of repeat num_envs times in the first dimension?

StoneT2000, 2024-12-19:
batch size 1 or repeated both should work, treat it like you are setting a batched torch tensor buffer representing the qpos

Try calling self.scene._gpu_apply_all() after setting qpos.

TritiumR, 2024-12-19:
It didn't work

StoneT2000, 2024-12-19:
Can run set qpos then apply all then run self.scene._gpu_fetch_all() then print the qpos? is it updated now?

TritiumR, 2024-12-19:
Thanks so much! It kind of works, the qpos are updated and the poses look similar but the qpos is not exactly the values I set. Is that normal in GPU sim?

TritiumR, 2024-12-19:
[image]

TritiumR, 2024-12-19:
Their is still small deviation, is there any way to make the deviation smaller?

StoneT2000, 2024-12-19:
Are you taking actions as well? And what controller are you using?

TritiumR, 2024-12-19:
Yes. I am using "pd_ee_delta_pose"

This is a pseudo code:
```
n = 30
for i in range(n)
        self.agent.robot.set_qpos(current_joint_angles) # (9,)
        self.env.scene._gpu_apply_all()
       
        obs, reward, terminated, truncated, info = self.env.step(actions) # (36, 7)

        current_joint_angles = self.agent.robot.get_qpos()[0]
```

StoneT2000, 2024-12-20:
Why does robot qpos need to be set? Is this because you are using CEM to explore action sequences then resetting back to a previous environment state to then take the best action?

TritiumR, 2024-12-20:
Exactly

StoneT2000, 2024-12-20:
Ok try using env.get_state_dict and env.set_state_dict (or get_state / set_state for flat tensors that you can batch and sample rows from)

You were only setting qpos doesn't cover joint velocity or object poses and velocities.

TritiumR, 2024-12-20:
Oh I see. It works perfectly right now. Thanks!

closed 2024-12-20 by TritiumR